rm-hull/luma.core #4: New scheme!

rogerdahl, 2017-01-15:
I like the new name :)

I'm wondering if splitting up the repo is the right thing to do though. I tend to like being able to grab everything with a single clone.

With a split repo, I'd suggest having a single repo called just "luma" that would pull in everything, with a "git clone --recursive".

rm-hull, 2017-01-15:
The motivation for splitting it up was to bring the **pcd8544** and **max7219** repos "into the fold" so that they could extend the core functionality. pcd8544 has already been done and renamed luma.lcd, max7219 will follow shortly and be renamed luma.led_matrix.

I didn't want one big repo with everything in: those repo's still get quite a bit of traffic (as does ssd1306), and there are hard links to them out on the net, so renaming them in github seemed like a good compromise (as github will redirect with a 301 for the old site).

I also wanted to encourage installing using the PyPi packages rather than git cloning - this seems to have been bourne out since cloning (excluding travis builders) has dropped right off since introducing PyPi packages.

Having said that, a meta repo "luma" sounds like it would work - are you suggesting adding the components as git submodules (or subtrees) ?

rogerdahl, 2017-01-15:
Thank you for the background info. Yes, I was thinking of submodules.

I didn't realize that the other new repos were from existing projects, but it's actually because `git clone` is not a good distribution method for end users that I think it's convenient to have everything in a single repo. So, anyone that wants to work on the library gets everything with a single `git clone`. Then, from the single repo, one can build any number of products for end users and make those available through distribution channels such as pip, other package managers or just GitHub Releases.

In a single repo approach, you could still keep the older repos in place in order to provide a link to the new home, and you could distribute stuff via Releases on the old repos.

It could be that I'm just being too lazy :) The motivation is simply to not have to put something together from multiple pieces in order to spin up a dev env for one of the components in the library and getting the test suite to run. I haven't actually tried working with the new project layout for luma yet, so I don't know how much of a concern it really is.

Of course, having a single repo also makes it so that there's only a single place for issues, PRs, etc, which may or may not be a good thing...

rogerdahl, 2017-01-15:
Just a bit more background -- around half a year ago, I moved a set of legacy SVN repos over to GH. The SVN repos were related but were distributed as separate packages on PyPI. When deciding how to handle them on GH, I found a couple of larger projects on GH that had merged their multiple repos, and some notes in favor of that approach. So I went ahead and merged the legacy repos while still distributing them as separate packages and I haven't noticed any disadvantages. Caveat emptor: I'm pretty much the only one that works on those packages :)

thijstriemstra, 2017-01-15:
The git commands to clone all sub-packages could be a script that lives on gist.github.com.

closed 2017-01-28 by thijstriemstra