rm-hull/luma.core #5: OSError: [Errno 5] Input/output error

thijstriemstra, 2017-01-15:
Was getting this traceback when there wasn't any I2C device connected to the RPi:

```
3d_box.py: error: [Errno 5] Input/output error
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/pi/projects/pi-test/luma.core/luma/core/device.py", line 70, in cleanup
    self.hide()
  File "/home/pi/projects/pi-test/luma.core/luma/core/device.py", line 53, in hide
    self.command(self._const.DISPLAYOFF)
  File "/home/pi/projects/pi-test/luma.core/luma/core/device.py", line 32, in command
    self._serial_interface.command(*cmd)
  File "/home/pi/projects/pi-test/luma.core/luma/core/serial.py", line 63, in command
    self._bus.write_i2c_block_data(self._addr, self._cmd_mode, list(cmd))
  File "/home/pi/.virtualenvs/pi-test/lib/python3.4/site-packages/smbus2/smbus2.py", line 274, in write_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 5] Input/output error
```

I thought this `OSError` business was already fixed before; it should throw a devicenotfound error..

thijstriemstra, 2017-01-15:
Wasn't using master here but still.

```
$ git branch
  master
* optimize-display
```

rm-hull, 2017-01-16:
Agree about the devicenotfounderror, but interesting in that:
* it reports the caught error: `3d_box.py: error: [Errno 5] Input/output error`
* the atexit function tries to do some cleanup, which if not initialized, also fails.

Possibly the simple 'hack' to the second point would be to wrap the atexit handler in a try/except/pass - after all we are in the process of exiting.

thijstriemstra, 2017-03-07:
With `serial-error` branch this now shows:

```
$ python examples/3d_box.py -i i2c
usage: 3d_box.py [-h] [--config CONFIG]
                 [--display {ssd1306,ssd1322,ssd1325,ssd1331,sh1106,pcd8544,st7735,max7219,neopixel}]
                 [--width WIDTH] [--height HEIGHT] [--rotate {0,1,2,3}]
                 [--interface {i2c,spi}] [--i2c-port I2C_PORT]
                 [--i2c-address I2C_ADDRESS] [--spi-port SPI_PORT]
                 [--spi-device SPI_DEVICE] [--spi-bus-speed SPI_BUS_SPEED]
                 [--bcm-data-command BCM_DATA_COMMAND] [--bcm-reset BCM_RESET]
                 [--bcm-backlight BCM_BACKLIGHT]
                 [--block-orientation {horizontal,vertical}]
                 [--mode {1,RGB,RGBA}]
                 [--framebuffer {diff_to_previous,full_frame}]
                 [--bgr {True,False}]
3d_box.py: error: I2C device not found on address: 60
```

PR #43 (2017-03-08): Catch input/output errors
fixes #5, #29
closed 2017-03-08 by rm-hull