rm-hull/luma.core #8: Multiple I2C devices

thijstriemstra, 2017-01-19:
PCF8574 allows you to use multiple I2C devices ([example](http://www.rototron.info/using-an-i2c-lcd-display-with-a-raspberry-pi/)). Is this something that can be supported?

rm-hull, 2017-01-19:
It is possible, as long as the I2C devices each have a unique I2C address. Sometimes there are 2 or 3 zero-ohm resisters which allow you to set the address, so you can run 3 of the same wthout clashes.

Same goes for SPI .. it is possible to support 2 devices per channel I think

thijstriemstra, 2017-01-23:
Apparently Adafruit created [a special class](https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/PCF8574.py) for this.

rm-hull, 2017-01-23:
Ah, right, gotcha - the 8574 is just an 8-bit IO expander, and that code is using it to serially clock data/commands into the 8574, and then latch the 8 bits all at once onto the parallel interface of a 1602 display - basically treating it as a glorified shift register.

They are often used to piggy back these displays: http://www.ebay.co.uk/itm/Serial-IIC-I2C-TWI-1602-16x2-LCD-Display-Module-for-Arduino-Raspberry-PI-/282269918634

[image]

This picture of the backside shows the tiny addon board (in black) with a 8574.

I had used a 1602 display coupled up to an MCP23017 (also I2C, but 16 I/O ports) a couple of years ago.

thijstriemstra, 2017-01-23:
> serially clock data/commands into the 8574, and then latch the 8 bits all at once onto the parallel interface of a 1602 display 

Ah simple... :-O I'm gonna start with this and see how it goes ;) http://www.rototron.info/using-an-i2c-lcd-display-with-a-raspberry-pi/

rm-hull, 2017-03-02:
These chips look interesting:
http://www.linear.com/product/LTC4316
http://www.linear.com/product/LTC4317
http://www.linear.com/product/LTC4318

So you could connect, say, two identical I2C-based SSD1306 128x64 screens to an LTC4317, and although they have the same address, the chip does address translation so they appear at addresses 0x3C and 0x3E

rm-hull, 2017-06-15:
@thijstriemstra - please see updated comments on #55 
Can this issue be closed?

thijstriemstra, 2017-06-16:
Yep.

closed 2017-06-16 by thijstriemstra