rm-hull/luma.core #29: OSError: [Errno 121] Remote I/O error

thijstriemstra, 2017-02-26:
Getting this error with the examples on rpi2:

```
$ python examples/3d_box.py -i i2c --display sh1106
2017-02-27 00:12:45,674 - [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "/home/pi/projects/pi-test/luma.examples/examples/demo_opts.py", line 87, in <module>
    rotate=args.rotate, mode=args.mode)
  File "/home/pi/projects/pi-test/luma.oled/luma/oled/device.py", line 70, in __init__
    self._const.CHARGEPUMP,         0x14)
  File "/home/pi/projects/pi-test/luma.core/luma/core/device.py", line 39, in command
    self._serial_interface.command(*cmd)
  File "/home/pi/projects/pi-test/luma.core/luma/core/serial.py", line 63, in command
    self._bus.write_i2c_block_data(self._addr, self._cmd_mode, list(cmd))
  File "/home/pi/.virtualenvs/pi-test/lib/python3.5/site-packages/smbus2/smbus2.py", line 274, in write_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error
```

I have not hooked up any i2c sh1106 device.

rm-hull, 2017-02-26:
That is the error I would expect smbus to throw because there is nothing at the 'remote' address (whatever the port value in `self._addr` is). Are you suggesting that this should be wrapped/captured & rethrown in a luma.core.error instead?

thijstriemstra, 2017-02-27:
It's very similar to #5 (same code path) but a different error. I suggest we catch them and throw appropriate core.errors.

thijstriemstra, 2017-02-27:
@rm-hull so this is basically (always) a `DeviceAddressError`?

thijstriemstra, 2017-03-06:
@kplindegaard any idea how we should treat this error?

kplindegaard, 2017-03-07:
@thijstriemstra I haven't really put much emphasis on catching anything from ioctl and possible exceptions it may throw if it's unable to perform the prescribed action as you understand. But yeah, this is typically what you experience when there is no device present on that given i2c address. I mean, I have sent illegal buffers totally unrecognizable for the slave device, but that's a different story. Whereas this is "device not available on this address" type issue.

thijstriemstra, 2017-03-07:
Thanks for the feedback @kplindegaard!

thijstriemstra, 2017-03-07:
Added `serial-error` branch and output is now:

```
$ python examples/3d_box.py -i i2c --display sh1106
usage: 3d_box.py [-h] [--config CONFIG]
                 [--display {ssd1306,ssd1322,ssd1325,ssd1331,sh1106,pcd8544,st7735,max7219,neopixel}]
                 [--width WIDTH] [--height HEIGHT] [--rotate {0,1,2,3}]
                 [--interface {i2c,spi}] [--i2c-port I2C_PORT]
                 [--i2c-address I2C_ADDRESS] [--spi-port SPI_PORT]
                 [--spi-device SPI_DEVICE] [--spi-bus-speed SPI_BUS_SPEED]
                 [--bcm-data-command BCM_DATA_COMMAND] [--bcm-reset BCM_RESET]
                 [--bcm-backlight BCM_BACKLIGHT]
                 [--block-orientation {horizontal,vertical}]
                 [--mode {1,RGB,RGBA}]
                 [--framebuffer {diff_to_previous,full_frame}]
                 [--bgr {True,False}]
3d_box.py: error: I2C device not found on address: 60
```

thijstriemstra, 2017-03-08:
fixed since 0.5.2

closed 2017-03-08 by thijstriemstra