rm-hull/luma.core #42: Add word-wrap capability to luma.core.virtual.terminal

rm-hull, 2017-03-06:
Currently, the terminal class will check that text to be rendered using proportional fonts wont overflow over the right hand edge. If there is not enough width to display the next character, it will insert a newline. For example:

[image]

Add a constructor arg setting so that wrapping works at the word level, eg, the above string would render as

```
The quick brown fox
jumps over the lazy
dog
```

thijstriemstra, 2017-03-13:
https://docs.python.org/3/library/textwrap.html looks useful here.

thijstriemstra, 2017-04-16:
Should be fixed with #78:

[image]

PR #78 (2017-04-17): add word wrap capability to luma.core.virtual.terminal
* fixes #42 
* adds `luma.core.interface` package (#41)
* move `luma.core.serial` to `luma.core.interface.serial`
* deprecate `luma.core.serial` module
closed 2017-04-17 by thijstriemstra