rm-hull/luma.core #44: Tests failing on RPi

thijstriemstra, 2017-03-08:
Some tests are failing on an RPI2 that has i2c/spi enabled.

thijstriemstra, 2017-03-08:
Output:

```
platform linux -- Python 3.5.3, pytest-3.0.6, py-1.4.32, pluggy-0.4.0 -- /home/pi/projects/pi-test/luma.core/.tox/py35/bin/python3.5
cachedir: .cache
rootdir: /home/pi/projects/pi-test/luma.core, inifile: 
plugins: cov-2.4.0
collected 70 items 

tests/test_dummy.py::test_capture_noops PASSED
tests/test_dummy.py::test_portrait PASSED
tests/test_dummy.py::test_dither PASSED
tests/test_framebuffer.py::test_full_frame PASSED
tests/test_framebuffer.py::test_diff_to_previous PASSED
tests/test_framerate_regulator.py::test_init_default PASSED
tests/test_framerate_regulator.py::test_init_unlimited PASSED
tests/test_framerate_regulator.py::test_init_30fps PASSED
tests/test_framerate_regulator.py::test_sleep PASSED
tests/test_framerate_regulator.py::test_effective_FPS PASSED
tests/test_framerate_regulator.py::test_average_transit_time PASSED
tests/test_history.py::test_restore_throws_error_when_empty PASSED
tests/test_history.py::test_save_and_restore_reverts_image PASSED
tests/test_history.py::test_drop_and_restore PASSED
tests/test_legacy.py::test_textsize PASSED
tests/test_legacy.py::test_text_space PASSED
tests/test_legacy.py::test_text_char PASSED
tests/test_legacy.py::test_show_message PASSED
tests/test_lib.py::test_rpio_gpio PASSED
tests/test_lib.py::test_spidev PASSED
tests/test_lib.py::test_multi PASSED
tests/test_mixin.py::test_display_not_implemented PASSED
tests/test_proportional.py::test_narrow_char PASSED
tests/test_proportional.py::test_wide_char PASSED
tests/test_proportional.py::test_space_char PASSED
tests/test_proportional.py::test_doublequote_char PASSED
tests/test_proportional.py::test_trim_not_nonzero PASSED
tests/test_serial.py::test_i2c_init_device_not_found PASSED
tests/test_serial.py::test_i2c_init_device_permission_error FAILED
tests/test_serial.py::test_i2c_init_device_address_error PASSED
tests/test_serial.py::test_i2c_init_no_bus PASSED
tests/test_serial.py::test_i2c_init_bus_provided PASSED
tests/test_serial.py::test_i2c_command PASSED
tests/test_serial.py::test_i2c_command_device_not_found_error PASSED
tests/test_serial.py::test_i2c_data PASSED
tests/test_serial.py::test_i2c_data_chunked PASSED
tests/test_serial.py::test_i2c_cleanup PASSED
tests/test_serial.py::test_spi_init PASSED
tests/test_serial.py::test_spi_init_invalid_bus_speed PASSED
tests/test_serial.py::test_spi_command PASSED
tests/test_serial.py::test_spi_data PASSED
tests/test_serial.py::test_spi_cleanup PASSED
tests/test_serial.py::test_spi_init_device_not_found FAILED
tests/test_serial.py::test_spi_unsupported_gpio_platform FAILED
tests/test_spritesheet.py::test_init PASSED
tests/test_spritesheet.py::test_len PASSED
tests/test_spritesheet.py::test_caching PASSED
tests/test_spritesheet.py::test_get PASSED
tests/test_spritesheet.py::test_get_string PASSED
tests/test_spritesheet.py::test_get_outofrange PASSED
tests/test_spritesheet.py::test_animate_unknown_seq PASSED
tests/test_spritesheet.py::test_animate_finite_seq PASSED
tests/test_spritesheet.py::test_animate_slow_seq PASSED
tests/test_spritesheet.py::test_animate_infinite_seq PASSED
tests/test_spritesheet.py::test_animate_subroutine PASSED
tests/test_terminal.py::test_wrapped_text PASSED
tests/test_terminal.py::test_tab_alignment PASSED
tests/test_terminal.py::test_control_chars PASSED
tests/test_terminal.py::test_scrolling PASSED
tests/test_terminal.py::test_alt_colors PASSED
tests/test_viewport.py::test_range_overlap_over12 PASSED
tests/test_viewport.py::test_range_overlap_over123 PASSED
tests/test_viewport.py::test_range_overlap_over23 PASSED
tests/test_viewport.py::test_range_overlap_over234 PASSED
tests/test_viewport.py::test_range_overlap_over34 PASSED
tests/test_viewport.py::test_range_overlap_over4 PASSED
tests/test_viewport.py::test_range_overlap_over_none PASSED
tests/test_viewport.py::test_snapshot_last_updated PASSED
tests/test_viewport.py::test_viewport_set_position PASSED
tests/test_viewport.py::test_viewport_hotspot PASSED

=================================== FAILURES ===================================
____________________ test_i2c_init_device_permission_error _____________________

    def test_i2c_init_device_permission_error():
        with pytest.raises(luma.core.error.DevicePermissionError) as ex:
>           i2c()
E           Failed: DID NOT RAISE <class 'luma.core.error.DevicePermissionError'>

tests/test_serial.py:58: Failed
________________________ test_spi_init_device_not_found ________________________

    def test_spi_init_device_not_found():
        import spidev
        with pytest.raises(luma.core.error.DeviceNotFoundError) as ex:
>           spi(gpio=gpio, spi=spidev.SpiDev())
E           Failed: DID NOT RAISE <class 'luma.core.error.DeviceNotFoundError'>

tests/test_serial.py:179: Failed
______________________ test_spi_unsupported_gpio_platform ______________________

    def test_spi_unsupported_gpio_platform():
        with pytest.raises(luma.core.error.UnsupportedPlatform) as ex:
>           spi(spi=spidev, port=9, device=1)
E           Failed: DID NOT RAISE <class 'luma.core.error.UnsupportedPlatform'>

tests/test_serial.py:185: Failed
----------------------------- Captured stderr call -----------------------------
./luma/core/serial.py:168: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(self._bcm_DC, self._gpio.OUT)
./luma/core/serial.py:169: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(self._bcm_RST, self._gpio.OUT)
===================== 3 failed, 67 passed in 10.12 seconds =====================
```

thijstriemstra, 2017-03-08:
So:

* UnsupportedPlatform: the RPi is obviously supported so we need a utility function that can do that (and skip the test accordingly)
* DeviceNotFoundError: skip if device is found or supply a value for a device that can never be found
* DevicePermissionError: skip if permission is there?

rm-hull, 2017-03-08:
For 

```
    def test_i2c_init_device_permission_error():
        with pytest.raises(luma.core.error.DevicePermissionError) as ex:
>           i2c()
E           Failed: DID NOT RAISE <class 'luma.core.error.DevicePermissionError'>
```

You probably want to call `i2c(port=423)` otherwise it will default to port 1 which will exist on the RPi

rm-hull, 2017-03-08:
For

```
    def test_spi_init_device_not_found():
        import spidev
        with pytest.raises(luma.core.error.DeviceNotFoundError) as ex:
>           spi(gpio=gpio, spi=spidev.SpiDev())
E           Failed: DID NOT RAISE <class 'luma.core.error.DeviceNotFoundError'>
```

Same deal, set the port rather than defaulting to 0

thijstriemstra, 2017-03-08:
Sounds good, will do.

thijstriemstra, 2017-03-08:
The permission error won't be triggered because theres no device on port 423, resulting in a device not found error.

PR #47 (2017-03-08): fix tests for rpi
fixes #44
closed 2017-03-08 by rm-hull