rm-hull/luma.core #52: support CHIP_IO (Next Thing C.H.I.P.)

combs, 2017-03-14:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a 
> documentation fix, or suggested improvement, please get in touch.

#### Type of Raspberry Pi 
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next. 
> This library has been tested on every variant except the RPi3.

Next Thing C.H.I.P.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

```Linux compy386 4.4.13-ntc-mlc #1 SMP Tue Dec 6 21:38:00 UTC 2016 armv7l GNU/Linux```

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen. 
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

Would love to see support for [CHIP_IO](https://github.com/xtacocorex/CHIP_IO), the C.H.I.P. version of RPi.GPIO :) I bet it would be just a few lines in `lib.py` and `serial.py`. (I will try this out myself but I am pretty noob.)

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the 
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

```
Traceback (most recent call last):
  File "st7735.py", line 14, in <module>
    serial = spi(port=0, device=0, bcm_DC=23, bcm_RST=24)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/serial.py", line 163, in __init__
    self._gpio = gpio or self.__rpi_gpio__()
  File "/usr/local/lib/python2.7/dist-packages/luma/core/lib.py", line 23, in __rpi_gpio__
    import RPi.GPIO
ImportError: No module named RPi.GPIO
```

combs, 2017-03-14:
Ha, yeah, just changing two lines in `lib.py` got me up and running—
```
    try:
         import CHIP_IO.GPIO
         return CHIP_IO.GPIO
#        import RPi.GPIO
#        return RPi.GPIO
```
(and using appropriate pins:
`serial = spi(port=32766, device=0, bcm_DC="CSID1", bcm_RST="CSID0")	`

thijstriemstra, 2017-03-14:
@combs instead of messing with the code you could also try (untested):

```
import CHIP_IO.GPIO
serial = spi(port=32766, device=0, gpio=CHIP_IO.GPIO, bcm_DC="CSID1", bcm_RST="CSID0")
```

rm-hull, 2017-03-14:
@thijstriemstra is right - you should be able to override the RPi.GPIO implementation exactly has described. 

Fantastic news that it works on CHIP with very little effort! 

I'd be interested to hear what the FPS you attain with the [perfloop.py](https://github.com/rm-hull/luma.examples/blob/master/examples/perfloop.py) example ... we will probably need to add some fixes to `demo_opts.py` to make this work with CHIP_IO though.

combs, 2017-03-15:
Yes! That worked a treat. Thank you @thijstriemstra.

@rm-hull `perfloop.py` is yacking out a bit. It expected each of the GPIO params to be integers, though in CHIP_IO we can use aliases (`CSID1` vs `133`) as futureproofing against kernel pin numbering changes

```
chip@chip:~/git/luma.examples/examples$ python perfloop.py --config ../conf/st7735.conf 
Testing dislay rendering performance
Press Ctrl-C to abort test

Traceback (most recent call last):
  File "perfloop.py", line 55, in <module>
    main()
  File "perfloop.py", line 28, in main
    device = get_device()
  File "/home/chip/git/luma.examples/examples/demo_opts.py", line 149, in get_device
    luma.lcd.aux.backlight(gpio_LIGHT=args.gpio_backlight).enable(True)
chip@chip:~/git/luma.examples/examples$ cat ../conf/st7735.conf 
--display=st7735
--interface=spi
--spi-bus-speed=32000000
--gpio-reset=132
--gpio-data-command=133
--gpio-backlight=134
--width=160
--height=128
```

I hacked it by manually adding `device` initializers and so on with the following results:
```
#### iter =  30411: render time = 1.69 ms, frame rate = 571.93 FPS
```

rm-hull, 2017-03-15:
ah, so that framerate is a bit artificial for the default st7735 implementation, it uses a `diff_to_previous` framebuffer and because the same image is rendered over and over, there is nothing new to display, and hence the very high 

If you add a `--framebuffer=fullframe` to your conf file it should be a bit more realistic

combs, 2017-03-15:
Hee hee, okay, that explains a lot 😄 

oof,
```
#### iter =    651: render time = 103.80 ms, frame rate = 9.63 FPS
```

closed 2017-04-09 by rm-hull