rm-hull/luma.core #54: Core docs

thijstriemstra, 2017-03-15:
Currently the same instructions to enable spi/i2c are duplicated across the various luma sub-projects. I think common docs like how to enable spi should be in the core docs, and device specific installation instructions belong in the specific sub-project docs.

rm-hull, 2017-03-15:
Ok, maybe now is the right time to have a single docs site?  ... https://github.com/rm-hull/luma.docs

have added you as a collaborator ;-)

thijstriemstra, 2017-03-15:
I guess it only needs a rtd site, no pypi stuff.

rm-hull, 2017-03-15:
Yes think so

rm-hull, 2017-03-15:
So, I just (mainly) copy & pasted the core `docs/` directory and created the rtd config - see https://luma-docs.readthedocs.io

Perhaps ought to think how to best structure the docs, but maybe document ideas & thoughts as issues in that repo.

hence, closing -this- issue

closed 2017-03-15 by rm-hull