rm-hull/luma.core #55: Some beginner questions

ski7777, 2017-03-15:
I am planning to run multiple I²C OLEDs with a Linux-based-Microcomputer which has two I²C interfaces. Now I have some Questions:

- How to enable I²C for example on this diplay? http://www.ebay.de/itm/0-96-I2C-IIC-SPI-Serial-128X64-OLED-LCD-Display-SSD1306-for-Arduino-51-STM32-/272475961798?hash=item3f70d539c6:g:EyUAAOSwo4pYRmeB
- How to select the I²C bus to be used?
- Is it possible to use multiple displays with the same address using an I²C multiplaxer like http://dsscircuits.com/sale/product/dssc0107 ?

Raphael

rm-hull, 2017-03-15:
> How to enable I²C for example on this diplay? http://www.ebay.de/itm/0-96-I2C-IIC-SPI-Serial-128X64-OLED-LCD-Display-SSD1306-for-Arduino-51-STM32-/272475961798?hash=item3f70d539c6:g:EyUAAOSwo4pYRmeB

There are some devices that are switchable between SPI and I2C - that listing looks to be one of them - if you look at the reverse side, it shows that if you want to use I2C you would need to short across R1, R4, R6, R7 and R8 to put it in I2C mode

[image]

You would be better off sourcing one that does I2C by default. @thijstriemstra (I think) had a device like this, and AFAIK ended up using it in 4-wire SPI mode, rather than soldering the open connections.

> How to select the I²C bus to be used?

http://luma-oled.readthedocs.io/en/latest/python-usage.html show you how you would configure I2C. If that documentation is not clear, then please raise a separate issue, and we'll try and clarify the text and examples.

> Is it possible to use multiple displays with the same address using an I²C multiplaxer like http://dsscircuits.com/sale/product/dssc0107 ?

It should be possible (actually that device looks quite useful), alternatively have a look at the discussion in issue https://github.com/rm-hull/luma.core/issues/8 - using an LTC431x should mean that two identically addressed devices appear at separate addresses, rather than having to manually multiplex them

ski7777, 2017-03-15:
Thank you fir the information.

Where would be the problem soldering a bypass to the resistors or is there a hardware problem?

Just switching between two or more displays before​ drawing new information would be possible but does the lib try to update the images periodically? If yes, maybe there I could try to write a module to switch across multiple displays automatically in the right moment. Maybe somebody could then write something like this for a I²C or SPI port expander to set the SPI CS.

Raphael

thijstriemstra, 2017-03-15:
Looks related to #8

rm-hull, 2017-03-15:
@ski7777 what device are you planning to use?

Raspberry Pi's with the 40-pin headers have 2x I2C buses: 
* port 1 on pins BCM_2 & BCM_3, and 
* port 0 on BCM_0 and BCM_1

You should be able to create 2 devices as follows (untested, but _should_ work):

```python

device0 = ssd1306(i2c(port=0, address=0x3C))
device1 = ssd1306(i2c(port=1, address=0x3C))

with canvas(device0) as draw:
    text.draw((0,0), text="BCM_0 and BCM_1", fill="white")

with canvas(device1) as draw:
    text.draw((0,0), text="BCM_2 and BCM_2", fill="white")
```

ski7777, 2017-03-15:
But here I have just one of two I²C bus which I can access so I need to
multiplex. So my question is: Is the display updated periodically via I²C?

Raphael

rm-hull, 2017-03-15:
If you were able to source one of the LTC431x multiplexers, I think the clearest thing would be to create two devices (the address of daisychained one would be 2 higher than the other address), so something like:

```python
device0 = ssd1306(i2c(port=1, address=0x3C))
device1 = ssd1306(i2c(port=1, address=0x3F))

with canvas(device0) as draw:
    text.draw((0,0), text="original", fill="white")

with canvas(device1) as draw:
    text.draw((0,0), text="daisychained", fill="white")
```

---
If you were to source one of the 4 channel multiplexers, you would have flip A0 (lets say it's connected to BCM_17) and then flip-flop the display:

[image]

```python
device = ssd1306(i2c(port=1, address=0x3C))

GPIO.output(17,  GPIO.LOW)

with canvas(device) as draw:
    text.draw((0,0), text="original", fill="white")

GPIO.output(17,  GPIO.HIGH)

with canvas(device) as draw:
    text.draw((0,0), text="daisychained", fill="white")
```

To answer your question:

> Is the display updated periodically via I²C?

Once a rendered image is flushed to the device it will remain persistent on the screen like that until a new image is rendered, or the program ends, so it is not necessary to refresh

rm-hull, 2017-03-15:
Just ordered one of these: http://www.ebay.co.uk/itm/191927201165, so will update this thread with further details when it arrives

ski7777, 2017-03-15:
That's the big brother of the one from dsscircuits which I have.

ski7777, 2017-03-16:
I just ordered the same breakout board

ski7777, 2017-03-16:
After a quick view in the source I think, it should be no problem to use
one i2c object for multiple displays, so I hope it will be possible to run
multiple displays at one multiplexer without using an extra GPIO pin.

rm-hull, 2017-03-16:
Yes, it should be possible to supply the same i2c interface to two different displays. I misread how the multiplexer works when I glanced at it initially - I'd assumed you needed to flip the A0-A2 bits, but that is the address offset - all the multiplexing is done by sending commands to a particular address. It shouldnt need any extra GPIO pins.

ski7777, 2017-03-16:
A multiplexer is just a cool array of MOSFETs controlled via I²C. I just
bought two of them: http://www.ebay.de/itm/272475961798

thijstriemstra, 2017-03-22:
Can this be closed?

ski7777, 2017-03-22:
I would like to keep this open until the parts delivered and I tested it.

Raphael

ski7777, 2017-03-31:
Some Minutes ago the Multiplexer arrived, but I am still wating for the OLEDs.
[image]

ski7777, 2017-04-02:
Yesterday the OLEDs arrived. I will test them as soon as possible.

Raphael

ski7777, 2017-04-02:
Just tested one display and the multiplexer but both did not respond. Trying to do
```
from luma.core.serial import i2c, spi
from luma.core.render import canvas
from luma.oled.device import ssd1306, ssd1325, ssd1331, sh1106

serial = i2c(port=1, address=0x3C)
device = ssd1306(serial)
```

ends up in
```
Traceback (most recent call last):
  File "/home/ftc/oled/luma/core/serial.py", line 86, in command
    list(cmd))
  File "/home/ftc/oled/smbus2/smbus2.py", line 274, in write_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ftc/oled/luma/oled/device.py", line 156, in __init__
    self._const.NORMALDISPLAY)
  File "/home/ftc/oled/luma/core/device.py", line 39, in command
    self._serial_interface.command(*cmd)
  File "/home/ftc/oled/luma/core/serial.py", line 91, in command
    'I2C device not found on address: {}'.format(self._addr))
luma.core.error.DeviceNotFoundError: I2C device not found on address: 60
```

My I²C Master has integrated 4k7 Pull-Up Resistors and my cables are not longer than 20cm.
Can somebody help me?

Raphael

ski7777, 2017-04-02:
P.S. Other Devices like my 4 Channel Multiplexer, the Wii Nunchuck and my BMP180 are working probably.

Raphael

rm-hull, 2017-04-02:
First thing I would suggest is to wire up a single OLED, and run `i2cdetect -y 1` on the command line to make sure you have a sane baseline result. Then try to run the `examples/demo.py` if you hadn't already.

Then I would put the multiplexer on the same bus, and check that it `i2cdetect` detects both that and the OLED. You should experiment to make sure the multiplexer is responding to commands properly.

Then I think what will need to happen is that the low level smbus will need to be initialized outside the library, then you send the control codes to address 0x70 (wherever the multiplexer is) and then create the `serial` object using the smbus object, 

e.g.

```python
smbus = smbus2.SMBus(port)
serial = i2c(bus=smbus, address=0x3C)
device = ssd1306(serial)
```

ski7777, 2017-04-02:
Both devices are not detected using `i2cdetect`: 
```
$ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
I did not try any example yet, because I don't think that there will be any difference . Currently I have just connected the OLED. 
Trying to initialize the bus by using
```
smbus = smbus2.SMBus(port)
serial = i2c(bus=smbus, address=0x3C)
device = ssd1306(serial)
```
does not help.

Raphael

rm-hull, 2017-04-02:
Is it a raspberry Pi you're using? (port 1 is the default for RPi)
If you are using a different type of single board computer, the ports may be configured differently. Try listing the ports with `ls -l /dev/i2c*`, and for each entry, repeat the i2cdetect command with the relevant number.

If you dont see anything with i2cdetect, then the luma library wont either.

ski7777, 2017-04-02:
It's not a raspberry Pi but it is definitely /dev/i2c-1

ski7777, 2017-05-09:
I could neither connect my oleds to my device nor to an arduino with adafruit test sketches :((

rm-hull, 2017-05-10:
I've been a bit sidetracked with work lately, but now have 2x SSD1306 and one of the multiplexers so will try and show a working demo in the next week or so

rm-hull, 2017-06-15:
### Implementation Notes
Hello, I finally got around to wiring this up, heres a picture and some demo code:

[image]

Blue wires are data and clock, black is ground, and orange is power.
I have wired up the ports on TCA9548A to the RPi as follows:

| TCA pin | Remarks | RPi pin | RPi function | |
|---|---|---|---|---|
| VIN | Power | P1-01 | 3V3 |
| GND | Ground | P1-6 | 0V |
| SDA | Serial Data | P1-03 | I2C1 SDA |
| SCL | Serial Clock | P1-05 | I2C1 SCL |
| RST | Reset | P1-01 | 3V3 | Via 10K pull-up resistor |
| SD0 | to SDA on display 0 | P1-01 | 3V3 | Via 10K pull-up resistor |
| SC0 | to SCL on display 0 | P1-01 | 3V3 | Via 10K pull-up resistor |
| SD1 | to SDA on display 1 | P1-01 | 3V3 | Via 10K pull-up resistor |
| SC1 | to SCL on display 1 | P1-01 | 3V3 | Via 10K pull-up resistor |

The pull-up resistors on the slave-side SDx/SCx pins are important ... I had some 10K's hanging around and they seemed ok. The RPi already has pull-ups (1K IIRC), so they are not needed there.

The crucial bits about the program below are:
* `mux_select` picks the correct display to select - this _must_ be done before each interaction with a specific device
* normally the luma device classes will initialize an `smbus2` object internally, but since we need access to it to control the TCA9548A we have to construct it manually and pass it into the SSD1306.
* the luma device classes normally clean-up on exit, but some extra code should be employed here to clear them down by `mux_select`ing and then calling cleanup manually (haven't bothered in the example).
   
```python
#!/usr/bin/env python
# -*- coding: utf-8 -*-
# Copyright (c) 2017 Richard Hull and contributors
# See LICENSE.rst for details.

"""
TCA9548A I2C Multiplexer example

Assumes the multiplexer is on address 0x70
"""
import time
import smbus2
from luma.core.interface.serial import i2c
from luma.oled.device import ssd1306
from luma.core.render import canvas

TCA_ADDR = 0x70

def mux_select(bus, tca_port):
    assert(0 <= tca_port <= 7)
    bus.write_byte(TCA_ADDR, 1 << tca_port)

def draw_stuff(device, count, message):
    with canvas(device) as draw:
        size = draw.textsize(message)
        x = device.width - size[0]
        draw.rectangle(device.bounding_box, outline="white")
        draw.text((10, 10), str(count), fill="white")
        draw.text((x // 2, 20), message, fill="white")

if __name__ == "__main__":
    port = 1  # set to 0 for original RPi or Orange Pi, etc
    bus = smbus2.SMBus(port)

    mux_select(bus, 0)
    device0 = ssd1306(serial=i2c(bus=bus, address=0x3C))

    mux_select(bus, 1)
    device1 = ssd1306(serial=i2c(bus=bus, address=0x3C))

    for count in range(300):
        # Select first screen
        mux_select(bus, 0)
        draw_stuff(device0, count, "Screen #0")
        time.sleep(2)

        # Select second screen
        mux_select(bus, 1)
        draw_stuff(device1, count, "Screen #1")
        time.sleep(2)
        draw_stuff(device1, count, "")
        time.sleep(2)

        mux_select(bus, 0)
        draw_stuff(device0, count, "")
        time.sleep(2)
```

In theory, we could connect up to 8 SSD1306 devices like this!

### References:
* http://www.ti.com/lit/ds/symlink/tca9548a.pdf
* http://www.ebay.co.uk/itm/191927201165

closed 2017-06-16 by rm-hull

ski7777, 2017-06-18:
My device has two 4k7 Pull-Ups but by directly attaching the Display or the Multiplexer i2cdetect does not find anything. Is this normal?

ski7777, 2017-06-18:
[image]
Left side: original display
Right side: my I²C soldering
Is this the right way or did I solder it the wrong way?

Raphael

rm-hull, 2017-06-18:
Hi, I think you may need to un-solder R3

rm-hull, 2017-06-18:
> My device has two 4k7 Pull-Ups but by directly attaching the Display or the Multiplexer i2cdetect does not find anything. Is this normal?

* What type of device are you using, and 
* What is the output of `ls -l /dev/i2c*`?