rm-hull/luma.core #61: Use pyfakefs to simulate /dev/i2c-1 device in tests

rm-hull, 2017-03-21:
* test_i2c_init_device_not_found
* test_i2c_init_device_permission_error

thijstriemstra, 2017-04-12:
smbus2 uses `os.open` to open the device, something that pyfakefs does not support: https://github.com/jmcgeheeiv/pyfakefs/blob/b7fc0b4d0c4d2a32197aa27ac48b30b2dca39f75/fake_filesystem_test.py#L867

thijstriemstra, 2017-04-12:
For example:

```
====================================================================== FAILURES =======================================================================
_____________________________________________________________ test_init_device_not_found ______________________________________________________________

    def test_init_device_not_found():
        from pyfakefs import fake_filesystem
        fs = fake_filesystem.FakeFilesystem()
        os_module = fake_filesystem.FakeOsModule(fs)
        pathname = '/dev/i2c-200'
        new_file = fs.CreateFile(pathname)
    
        port = 200
        address = 0x710
        with patch('smbus2.smbus2.os', os_module):
            with pytest.raises(luma.core.error.DeviceNotFoundError) as ex:
>               i2c(port=port, address=address)

tests/test_i2c.py:43: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
luma/core/serial.py:60: in __init__
    self._bus = bus or smbus2.SMBus(port)
.tox/py35/lib/python3.5/site-packages/smbus2/smbus2.py:132: in __init__
    self.open(bus)
.tox/py35/lib/python3.5/site-packages/smbus2/smbus2.py:142: in open
    self.fd = os.open("/dev/i2c-{}".format(bus), os.O_RDWR)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <pyfakefs.fake_filesystem.FakeOsModule object at 0x7fb266422cf8>, file_path = '/dev/i2c-200', flags = 2, mode = None

    def open(self, file_path, flags, mode=None):
        """Return the file descriptor for a FakeFile.
    
            WARNING: This implementation only implements creating a file. Please fill
            out the remainder for your needs.
    
            Args:
              file_path: the path to the file
              flags: low-level bits to indicate io operation
              mode: bits to define default permissions
    
            Returns:
              A file descriptor.
    
            Raises:
              OSError: if the path cannot be found
              ValueError: if invalid mode is given
              NotImplementedError: if an unsupported flag is passed in
            """
        if flags & os.O_CREAT:
            fake_file = FakeFileOpen(self.filesystem)(file_path, 'w')
            if mode:
                self.chmod(file_path, mode)
            return fake_file.fileno()
        else:
>           raise NotImplementedError('FakeOsModule.open')
E           NotImplementedError: FakeOsModule.open

.tox/py35/lib/python3.5/site-packages/pyfakefs/fake_filesystem.py:2595: NotImplementedError
```

thijstriemstra, 2017-04-12:
I'll create a wrapper for `FakeOsModule`.

PR #75 (2017-04-12): don't rely on i2c-tools in tests
fixes #61
commit 425f873b (2017-04-12): don't rely on i2c-tools in tests (#75)

fixes #61
closed 2017-04-12 by thijstriemstra