rm-hull/luma.core #90: failing tests on raspberry pi

thijstriemstra, 2017-07-03:
```
============================================= FAILURES ==============================================
_______________________________________ test_make_serial_spi ________________________________________

    def test_make_serial_spi():
        """
        :py:func:`luma.core.cmdline.make_serial.spi` returns an SPI instance.
        """
        factory = cmdline.make_serial(test_spi_opts)
        with pytest.raises(error.UnsupportedPlatform):
>           factory.spi()
E           Failed: DID NOT RAISE <class 'luma.core.error.UnsupportedPlatform'>

tests/test_cmdline.py:115: Failed
___________________________________ test_make_serial_spi_alt_gpio ___________________________________

    def test_make_serial_spi_alt_gpio():
        """
        :py:func:`luma.core.cmdline.make_serial.spi` returns an SPI instance
        when using an alternative GPIO implementation.
        """
        class opts(test_spi_opts):
            gpio = 'fake_gpio'
    
        with patch.dict('sys.modules', **{
                'fake_gpio': Mock(unsafe=True)
            }):
            factory = cmdline.make_serial(opts)
            with pytest.raises(error.DeviceNotFoundError):
>               factory.spi()
E               Failed: DID NOT RAISE <class 'luma.core.error.DeviceNotFoundError'>

tests/test_cmdline.py:131: Failed
______________________________________ test_create_device_oled ______________________________________

    def test_create_device_oled():
        """
        :py:func:`luma.core.cmdline.create_device` supports OLED displays.
        """
        display_name = 'oled1234'
        display_types = {'oled': [display_name]}
    
        class args(test_spi_opts):
            display = display_name
    
        module_mock = Mock()
        with patch.dict('sys.modules', **{
                'luma': module_mock,
                'luma.oled': module_mock,
                'luma.oled.device': module_mock
            }):
            with pytest.raises(error.UnsupportedPlatform):
>               cmdline.create_device(args, display_types=display_types)
E               Failed: DID NOT RAISE <class 'luma.core.error.UnsupportedPlatform'>

tests/test_cmdline.py:161: Failed
====================================== pytest-warning summary =======================================
WW0 in tests/test_cmdline.py:108 the following warning was recorded:
 luma/core/interface/serial.py:171: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)

WW0 in tests/test_cmdline.py:189 the following warning was recorded:
 luma/core/serial.py:18: DeprecationWarning: luma.core.serial is deprecated and will be removed in v1.0.0: use luma.core.interface.serial instead
  'use luma.core.interface.serial instead')

====================== 3 failed, 114 passed, 2 pytest-warnings in 9.78 seconds ======================
```

rm-hull, 2017-07-04:
Suspect that all three of these are using real spi or i2c (which is present on the pi, obvs), maybe they need to do something similar to `test_make_serial_spi_alt_gpio` that slots in a fake spi and i2c as well as the fake gpio.

TBH I'm having tremendous problems running the test on ax x86 linux machine at the moment:

```
=========================================================================== FAILURES ============================================================================
____________________________________________________________________ test_module_deprecated _____________________________________________________________________

    def test_module_deprecated():
        """
        The :py:mod:`luma.core.serial` module is deprecated.
        """
        msg = ("luma.core.serial is deprecated and will be removed in " +
            "v1.0.0: use luma.core.interface.serial instead")
    
>       with pytest.deprecated_call() as c:
E       TypeError: deprecated_call() takes at least 1 argument (0 given)

tests/test_serial.py:20: TypeError
__________________________________________________________________ test_init_params_deprecated __________________________________________________________________

    def test_init_params_deprecated():
        port = 5
        device = 2
        bus_speed = 16000000
        dc = 80
        rst = 90
        msg1 = 'bcm_DC argument is deprecated in favor of gpio_DC and will be removed in 1.0.0'
        msg2 = 'bcm_RST argument is deprecated in favor of gpio_RST and will be removed in 1.0.0'
    
>       with pytest.deprecated_call() as c:
E       TypeError: deprecated_call() takes at least 1 argument (0 given)

tests/test_spi.py:72: TypeError
_______________________________________________________________________ test_deprecation ________________________________________________________________________

    def test_deprecation():
        """
        ``deprecated`` creates a DeprecationWarning.
        """
        class DeprecatedClass(object):
            def __init__(self):
                self.msg = 'Deprecated; will be removed in 0.0.1'
                util.deprecation(self.msg)
    
>       with pytest.deprecated_call() as c:
E       TypeError: deprecated_call() takes at least 1 argument (0 given)

tests/test_util.py:30: TypeError
============================================================= 3 failed, 114 passed in 4.07 seconds ==============================================================
```

thijstriemstra, 2017-07-04:
Looks related to pytest?

rm-hull, 2017-07-06:
Something changed in pytest recently to do with how they handled deprecation warnings. Removed it for now, as these tests will be removed entirely when we hit 1.0.0

As to the failing tests on RPi: not sure - do we need to fake out the SPI and I2C ... possibly that is the reason they are throwing those errors on the Pi?

My typical workflow is to test the code on desktop linux as this is how the tests run in Travis (and besides, running tox on RPi is painfully slow). I tend to use `sshfs` to share the repo directory on the Pi - so I can run the actual code on the device and the tests on a desktop.

thijstriemstra, 2017-09-15:
Tried running the tests again on RPi2 with Python 3.4.3. Tox died with: `MemoryError` :)

Will try on RPi3 now.

thijstriemstra, 2017-09-15:
PR #101 fixes running tests on rpi. On regular linux, e.g. ubuntu, it'll skip when spi/i2c devices etc can't be found:

```
==== short test summary info ======
SKIP [1] luma.core/tests/test_cmdline.py:119: GPIO access not available
SKIP [1] luma.core/tests/test_cmdline.py:138: SPI device not found
SKIP [1] luma.core/tests/test_cmdline.py:174: GPIO access not available

 110 passed, 3 skipped in 4.10 seconds
```

thijstriemstra, 2017-09-15:
And same on Travis:

```
=========================== short test summary info ============================

SKIP [1] /home/travis/build/rm-hull/luma.core/tests/test_cmdline.py:119: GPIO access not available

SKIP [1] /home/travis/build/rm-hull/luma.core/tests/test_cmdline.py:138: SPI device not found

SKIP [1] /home/travis/build/rm-hull/luma.core/tests/test_cmdline.py:174: GPIO access not available
```

thijstriemstra, 2017-09-15:
@rm-hull so it's ok on linux, what about osx, dont have a machine around atm. Same test results?

closed 2017-09-16 by rm-hull

rm-hull, 2017-09-16:
Heres all the failed & skipped on osx:

```
tests/test_bitbang.py::test_unsupported_gpio_platform SKIPPED
tests/test_cmdline.py::test_make_serial_spi FAILED
tests/test_cmdline.py::test_make_serial_spi_alt_gpio FAILED
tests/test_cmdline.py::test_create_device_oled FAILED
tests/test_spi.py::test_init_device_not_found SKIPPED
tests/test_spi.py::test_unsupported_gpio_platform SKIPPED
===================================== short test summary info ======================================
SKIP [1] /Users/rhu/dev/luma.core/tests/test_bitbang.py:68: RPi.GPIO is not supported on this platform: Darwin
SKIP [1] /Users/rhu/dev/luma.core/tests/helpers.py:44: spidev is not supported on this platform: Darwin
SKIP [1] /Users/rhu/dev/luma.core/tests/test_spi.py:110: RPi.GPIO is not supported on this platform: Darwin

```

PR #101 (2017-09-16): fix running tests on rpi
fixes #90