rm-hull/luma.core #98: Travis builds on python 3.6 breaks.

rm-hull, 2017-09-04:
Have just experienced this on a branch, but is also happening on master (see https://travis-ci.org/rm-hull/luma.core/jobs/259461663#L313). This did build cleanly on master - it was green, restarted it, and now failing !

@thijstriemstra - any ideas/insight?

thijstriemstra, 2017-09-04:
Well there's this https://github.com/pypa/setuptools/issues/1086, looking into it.

rm-hull, 2017-09-04:
Maybe just upgrading travis's setuptools is the simple solution: https://github.com/labgrid-project/labgrid/pull/119/files

thijstriemstra, 2017-09-04:
We're already doing that.

thijstriemstra, 2017-09-04:
Your fix seems to have worked in #97.

rm-hull, 2017-09-04:
See https://github.com/rm-hull/luma.core/pull/97/commits/58d672825d25f60749b7f603bb16495097697d4f

commit d2218475 (2017-09-04): Conditionally install RPi.GPIO and spidev only on linux targets (not OSX/Darwin) (#97)

* Conditionally install RPi.GPIO and spidev only on linux targets (not OSX/Darwin)

This goes some way to remediate the issue described in https://github.com/rm-hull/luma.emulator/issues/18

* Use platform.system() instead
* Update changelog
* setuptools fix (fixes #98)
* Use expanded flags
* Cater for non-install on Windows too
closed 2017-09-04 by rm-hull