rm-hull/luma.core #109: Start signing git commits

thijstriemstra, 2017-10-31:
See https://help.github.com/articles/signing-commits-with-gpg/

rm-hull, 2017-11-01:
Curious how this will work if, say, I merge your branch and squash your signed commits via github's webpage?

thijstriemstra, 2017-11-01:
@rm-hull that's what brought this signing to my attention: when you squash using github a box appears saying 'verified' appears next to the commit because it's signed (using github's pgp key).

thijstriemstra, 2017-11-01:
[image]

rm-hull, 2017-11-01:
I just retagged luma.oled 2.3.0 

[image]

rm-hull, 2017-11-01:
Now to try and work out how to have one GPG key across all my computers ...

thijstriemstra, 2017-11-01:
I can image you can add multiple, just like SSH keys..

thijstriemstra, 2017-11-01:
And new contributors do not need to worry because we're the one merging (and signing).

thijstriemstra, 2017-11-04:
I've also enabled it.

closed 2017-11-04 by thijstriemstra