rm-hull/luma.core #113: [feature request] Sharp memory LCD

Stefal, 2017-11-25:
Hi !
Following my question for menu management #112 I'd like to know if it's possible to support the Sharp memory LCD. These low power LCDs are a good solution when you want something readable under direct [sunlight](https://www.youtube.com/watch?v=Qe2bWN900Yk), without the e-ink drawback (slow refresh rate).
I plan to use this lcd on my bike for collection photos for OpenStreetMap contributions (you can see more [here](https://twitter.com/i/moments/786549880365867008).

Sharp Memory LCD doc:
[datasheet](https://www.sharpsma.com/documents/1468207/1485747/Memory+LCD+Theory%2C+Programming%2C+and+Interfaces)
programming Memory LCD (https://cdn.sharpsde.com/fileadmin/products/Displays/2016_SDE_App_Note_for_Memory_LCD_programming_V1.3.pdf)
Some code on github:
https://github.com/MakerDyne/Memory-LCD-for-Raspberry-Pi

rm-hull, 2017-11-25:
Glancing at the datasheet, it appears to be an SPI device, so interfacing to it should be relatively straightforward. The sticking point to writing a driver is having one to test against though - are you able to source one for us ... either as a donation or a loan?

thijstriemstra, 2017-11-26:
This ticket should probably be moved to the luma.lcd repo at some point.

Stefal, 2017-11-26:
@rm-hull I have an old one, from [Adafruit](https://www.adafruit.com/product/1393). I will use it in the future, but in the meantime, I can send it to you. You're in the UK ? Give me your postal address and I will ship it.

rm-hull, 2017-11-27:
@Stefal - sent a mail to your email address from one of your repo’s commits with my details. Let me know if it didn't arrive.

rm-hull, 2017-12-14:
Closing - see linked issue: https://github.com/rm-hull/luma.lcd/issues/61 - updates/progress will be posted there instead

closed 2017-12-14 by rm-hull