rm-hull/luma.core #115: Support for Linux i2c

gertmenke, 2017-12-07:
Hi,
I have an ssd1306 OLED connected to my ThinkPad via i2c and would like to use Luma to access it. However, it seems that it needs RPi.GPIO which obviously makes little sense in my case.
Is there a way to use the standard i2c kernel module instead?

gertmenke, 2017-12-07:
Okay, never mind. I just found out that I can just install RPi.GPIO and not use it. Maybe it shouldn't necessarily be a dependency, though?
Anyway, sorry for asking a dumb question.

closed 2017-12-07 by gertmenke

rm-hull, 2017-12-07:
I think at some point we had it not as an explicit dependency, but it somehow got added along the way ... probably because it was needed for orchestrating with SPI, plus most of the user-base are RPi users, and RPi.GPIO works on some other SBCs.

I was always aware that I2C is available on linux generally, but curious to know how you (physically) expose I2C from your thinkpad - some close up pics of your wiring and the working artefact would be good!

 :)

thijstriemstra, 2017-12-08:
I thought @gertmenke was referring to https://www.raspberrypi.org/blog/gpio-expander/

rm-hull, 2017-12-08:
Awww man, that's awesome! Opens up a whole new avenue... Am definitely going to try that!

gertmenke, 2017-12-08:
Thanks! :-)

I'm also thinking of building a display directly into a VGA plug. That would be nice, especially when the Laptop is running with the lid closed. Unfortunately, most of the smaller displays have a fixed I2C address, so I couldn't use them with my modified ThinkPad.