rm-hull/luma.core #116: Terminology and luma.core.virtual.viewport

Stefal, 2017-12-18:
Hi !
As a beginner, I'm trying to understand luma.core, and I'm learning the various concepts and terminology used for GUI.
One thing suprised me: If I read the [viewport](https://en.wikipedia.org/wiki/Viewport)'s page on Wikipedia, it's not the same concept as I understand it in caroussel.py or crawl.py.
From wikipedia: 
> the viewport is the visible portion of a 2D area which is larger than the visualization device
But in caroussel.py, the viewport is larger than the visualization device.

Am I wrong ?

rm-hull, 2017-12-18:
Viewport as defined in wikipedia is correct, of course. In luma.core, if you treat `viewport` as a class then, in the conventional sense, yes i think it is mixing the terminology up. 

If you take a snippet as follows:

```python
    virtual = viewport(device, width=device.width, height=768)

    for _ in range(2):
        with canvas(virtual) as draw:
            draw.text((0, 0), "A long time ago", fill="white")
            draw.text((0, 12), "in a galaxy far", fill="white")
            draw.text((0, 24), "far away....", fill="white")
```

I read this as _"create a virtual device with a viewport onto the device, sized as ..."_.

Strictly speaking `virtual` is an instance of the `viewport` class, so I absolutely see where you're coming from, but it had never really occurred to me to read the code with that inflection before.

Stefal, 2017-12-18:
Ok

If I'm the only one understanding this way, it's not a problem.

rm-hull, 2017-12-18:
No no, it's good that you're picking stuff up like this. If we can make the documentation and the code clearer as a result, then it's better for everyone who comes to subsequently use it. Have a think how it could be changed to be consistent with the wikipedia definition.