rm-hull/luma.core #119: New Font: Spaces between characters

jonathanrjpereira, 2018-02-04:
When creating a new font, do I need to write a column 0x00 so that there is a space between any two characters? 

For example: Assume the number 3 is to be displayed in an area of 3 columns * 8 rows. Assuming it the 3 the bit pattern of the columns is 0x49, 0x49, 0x7F.

Is it necessary to add an additional 0x00 to create a space between any two characters? In such a case the bit patterns of the columns will then be 0x49, 0x49, 0x7F, 0x00.

From what I have observed, every character in the LCD_FONT has its last three columns 0x00 whereas TINY_FONT only has the last column as 0x00. 

For my project, I wanted to fit 8 numbers on a single 4in1 cascaded (8*8) matrix using static text. But none of the existing fonts were capable of doing so. TINY_FONT, in my opinion, wasn't looking big enough to read easily whilst optimizing the display area. So I'm currently trying to create a font that is similar to a 7 segment font wherein each character only consumes a fixed area of 3 columns & 7 rows. The 4th column will act as a space between the two characters. 
My reference font style is: [7 Segment Font](https://www.dafont.com/digital-7.fontl) 
The `7SEG_FONT` **_only_** supports Capital Letters(0x41 - 0x5A inclusive) & Numbers(0x30 - 0x39 inclusive). I haven't calculated the bit patterns for any other special characters since I feel that the purpose of this font is only for a few static display alpha-numeric applications.

rm-hull, 2018-02-04:
> Is it necessary to add an additional 0x00 to create a space between any two characters? In such a case the bit patterns of the columns will then be 0x49, 0x49, 0x7F, 0x00

I would say so, yes.

It would be good/useful if you were prepared to share your font creation by submitting a PR for inclusion in this library.

jonathanrjpereira, 2018-02-04:
I would love to share this font for inclusion in this library. 

Is it ok if I use the same bit patterns for both Uppercase & Lowercase alphabets?
Since the font which I am referring to does not support lowercase alphabets. This will make it easier for the user since they can type their strings in either Uppercase or Lowercase.

rm-hull, 2018-02-04:
Awesome.

Duplicating the upper case letters for the corresponding lower case letters is probably the simplest thing to do if they don't already exist.

thijstriemstra, 2018-02-05:
fixed with #120

closed 2018-02-05 by thijstriemstra