rm-hull/luma.core #122: [Question] UTF-8 support

shymmq, 2018-02-11:
When I try to draw a text containing a unicode character, I get the following error:
```
error from callback <function on_message at 0xb64edef0>: 'ascii' codec can't encode character u'\u015b' in position 17: ordinal not in range(128)
  File "/usr/local/lib/python2.7/dist-packages/websocket/_app.py", line 269, in _callback
    callback(self, *args)
  File "pushbullet.py", line 48, in on_message
    term.println(title)
  File "/pishare/src/luma.core/luma/core/virtual.py", line 268, in println
    self.puts(text)
  File "/pishare/src/luma.core/luma/core/virtual.py", line 282, in puts
    for method, args in ansi_color.find_directives(text, self):
  File "/pishare/src/luma.core/luma/core/ansi_color.py", line 103, in find_directives
    for directive in parse_str(text):
  File "/pishare/src/luma.core/luma/core/ansi_color.py", line 62, in parse_str
    text = str(text)
```

Are there any plans to support encodings other than ASCII? Or is that a limitation of the PIL library?

rm-hull, 2018-02-11:
Someone else reported this a few months ago: https://github.com/rm-hull/luma.oled/issues/161

So I should say Pillow does support Unicode fonts, but you are using the terminal class which doesn't presently support Unicode. I have PR (https://github.com/rm-hull/luma.core/pull/93) in which I intended to fix this but have neglected it for a few months. I will try and get it into a passing state and push to master

thijstriemstra, 2018-03-22:
@shymmq can you try the changes made in the pull request #93 and see if it fixes your problem? On Python 2.7 you might have to prefix the string with `u`, e.g.

```python
my_text = u"Текст на русском"
```

closed 2018-03-22 by rm-hull

shymmq, 2018-03-22:
Sure, I'll check it out once I have some free time. Thanks!

thijstriemstra, 2018-03-23:
Thanks, you can now use luma.core 1.7.1 or newer to test.

PR #93 (2018-03-22): Support accented characters
This PR will attempt to fix the regex bug and python 2 compatibility mentioned in issue https://github.com/rm-hull/luma.oled/issues/161

fixes #122