rm-hull/luma.core #124: Higher-level abstractions for setting SPI transfer_size

drspangle, 2018-02-20:
This is a feature request, not a bug report.

Currently, the command line parsers and device configuration settings for the MAX7219 devices in the luma library allow the SPI bus speed (i.e. clock speed) and CS active-high to be set, but there's no way to adjust the maximum `transfer_size` parameter, which is exposed by the `spi` class constructor (the default is 4096). I think this would be very useful for me to debug my glitchy MAX7219-based displays; the raspberry pi supports 4096, but I think the cascaded MAX7219s in my display would be happier with smaller words. I don't want to have to manually override the default parameter in the core library, it would be great if this could be supported from the command line argument parser.

drspangle, 2018-02-21:
@rm-hull Would I be correct to assume that you're going to implement this feature at some point in the near future? If there's anything I can do to help, please let me know.

thijstriemstra, 2018-02-21:
Maybe you can try it, it's exactly the same as https://github.com/rm-hull/luma.core/commit/5c688227fc26d0373a92e778813dc60fb514aeba but then for transfer_size @drspangle

drspangle, 2018-02-21:
@thijstriemstra Fair enough, I created a new PR #125 based on that commit you linked.

rm-hull, 2018-02-21:
I was going to, but you beat me to it

rm-hull, 2018-02-21:
Will create a new release tomorrow for this

closed 2018-02-21 by rm-hull