rm-hull/luma.core #129: SEG7_FONT up site down

DJJo14, 2018-03-27:
#### Type of Raspberry Pi
RPI1B (not plus)

#### Linux Kernel version
4.9.80+ #1098 Fri Mar 9 18:51:28 GMT 2018 armv6l GNU/Linux

#### Expected behaviour
Display is showing the time and is made at 19:12:35

[image]

#### Actual behaviour
the numbers 96752 are upsite down, does not happen with other fonts

### Running code
```
from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.led_matrix.device import max7219
from luma.core.legacy.font import proportional, LCD_FONT, CP437_FONT, TINY_FONT, SINCLAIR_FONT, SEG7_FONT
from luma.core.legacy import text
import time
import datetime

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, cascaded=5, block_orientation=-90)
while 1:
        with canvas(device) as draw:
                now = datetime.datetime.now()
                t = now.strftime('%H:%M:%S')
                text(draw, (0, 0), t, fill="white", font=(SEG7_FONT))
        time.sleep(0.1)

```

thijstriemstra, 2018-03-27:
@DJJo14 have you tried adjusting the `block_orientation` to a different value?

rm-hull, 2018-03-27:
I'll check locally with the emulator & write a little script to correct the values in the source code.

cc: @jonathanrjpereira

jonathanrjpereira, 2018-03-27:
@DJJo14 Try changing the `block_orientation` to either `0` or `90` instead of `-90`.

DJJo14, 2018-03-27:
thanks all for the fast replay, but it is not the block_orientation because when i change the font the orientation is correct.
[image]

rm-hull, 2018-03-27:
I tried on the emulator and can confirm SEG7_FONT is indeed upside down. I thought it probably was based on @DJJo14 comments:

> the numbers 96752 are upsite down, does not happen with other fonts

Fix forthcoming ...

PR #130 (2018-03-28): Reverse bit patterns for SEG7_FONT
Hacky script used to reverse bit patterns:

```python
#!/usr/bin/env python

from luma.core.legacy.font import SEG7_FONT

def transform(bit_pattern):
    a = bit_pattern >> 7 & 0x01
    b = bit_pattern >> 6 & 0x01
    c = bit_pattern >> 5 & 0x01
    d = bit_pattern >> 4 & 0x01
    e = bit_pattern >> 3 & 0x01
    f = bit_pattern >> 2 & 0x01
    g = bit_pattern >> 1 & 0x01
    h = bit_pattern >> 0 & 0x01
    return a | b << 1 | c << 2 | d << 3 | e << 4 | f << 5 | g << 6 | h << 7

for i, bitmaps in enumerate(SEG7_FONT):
    if 32 <= i < 127:
        ch = "'{0}'" [... cut]
closed 2018-03-28 by rm-hull