rm-hull/luma.core #141: Unsupported platform

ameel, 2018-07-05:
Hi, I'm trying to use this on a SBC with rk3399 chip (rock960). It has SPI port 32766, device 0. DC is at port gpio1006 (as shown in /sys/class/gpio/), and RST is at gpio1002 (as shown in /sys/class/gpio).

Despite 'configuring' it for my device, I get error "luma.core.error.UnsupportedPlatform: GPIO access not available." when I try to run anything. Is there any way to use the library for my device?

#### Type of Raspberry Pi
Not a raspberry pi. SBC uses a rk3399 chip

#### Linux Kernel version
Linux rock960 4.4.103 #2 SMP Mon Jan 15 11:30:25 CST 2018 aarch64 aarch64 aarch64 GNU/Linux

#### Expected behaviour
Library works

#### Actual behaviour
Error is raised: luma.core.error.UnsupportedPlatform: GPIO access not available.

thijstriemstra, 2018-07-19:
Is there a RPi.GPIO compatible library available for your platform?

thijstriemstra, 2019-01-07:
ping @ameel

closed 2019-01-07 by thijstriemstra