rm-hull/luma.core #148: Modifying attributes of 'Word_wrap'

Flantinori, 2018-08-28:
Hi there,
And thanks for creating this wonderful library.
I am using it to control an OLED 1351 and it works wonderfully on a Raspberry Pi Zero W.
At the moment I am using your terminal example and would like to modify the word_wrap, in order to control the width of the screen and make sure it breaks the words where I would like it to.
I am reading a string of text that I have split with r.text.split   

for line in allLines:
term = terminal(device, font, color="red", line_height=None, word_wrap=True)
                term.println(line)
                time.sleep(3)

Thank you so much for your help!

All the best

FA

rm-hull, 2018-08-29:
Have you tried just doing:

```python
term = terminal(device, font, color="red", word_wrap=True)
term.println(allLines)
time.sleep(3)
```

?

Flantinori, 2018-08-31:
Hey, thanks for helping.
I tried the code, but I receive the following message
`Traceback (most recent call last):
  File "ac_06_03.py", line 83, in <module>
    main()
  File "asc_06_03.py", line 69, in main
    term.println(allLines)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 246, in println
    directives = ansi_color.find_directives(text, self)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/ansi_color.py", line 102, in find_directives
    for directive in parse_str(text):
  File "/usr/local/lib/python2.7/dist-packages/luma/core/ansi_color.py", line 64, in parse_str
    result = prog.match(text)
TypeError: expected string or buffer`

Other than this, it has given me a good idea for something else.
Will keep working on it and will send some news soon.
Thank you!

thijstriemstra, 2018-12-22:
@Flantinori you can access the python `TextWrapper` that controls the word wrapping using the `tw` attribute of the `terminal` class. By default it's configured like this:

```python
self.tw = TextWrapper()
self.tw.width = self.width
self.tw.expand_tabs = False
self.tw.replace_whitespace = False
self.tw.drop_whitespace = False
self.tw.break_long_words = True
```

so try:

```python
term = terminal(device, font, color="red", word_wrap=True)

# for example:
term.tw.expand_tabs = True
term.tw.drop_whitespace = True

for line in allLines:
    term.println(line)
```

closed 2018-12-22 by thijstriemstra