rm-hull/luma.core #152: Support for MicroPython by using machine.i2c instead of python-smbus

danielmader, 2018-12-18:
Hello, I've recently started to work myself into ESP8266 boards and have come across MicroPython. I'd like to interface with a ST7567A based LCD in order to build a projection clock, but I'm stuck without a driver. [Thankfully, proper googling brought me to this project :)](https://forum.micropython.org/viewtopic.php?f=14&t=5663)

Is it possible to modify luma.core to make use of MicroPythons machine module? Also, I probably cannot use PIL due to the restricted hardware. Will it be possible to use luma.core/luma.lcd without PIL?

Thank you many times in advance, and please excuse if this is a dumb question. I'm a little overwhelmed, to be honest.

HW: [D-duino V2 (NodeMCU 1.0 ESP-12F ESP8266)](http://dong-sen.com/d-duino/) 
OS/env: [MicroPython 1.9.4](https://www.micropython.org/)

thijstriemstra, 2018-12-18:
@danielmader see https://github.com/rm-hull/luma.oled/issues/216

danielmader, 2018-12-18:
@thijstriemstra, thanks for linking this issue, seems I have looked in the wrong place. Sadly, the outlook for being able to use luma.lcd is not good, if I understood correctly...

closed 2020-09-20 by thijstriemstra