rm-hull/luma.core #155: Retrieve supported display modes for device

thijstriemstra, 2019-01-13:
Would be nice to have a unified API that returns the supported display modes for a certain device. Right now these values are hardcoded and impossible to retrieve, e.g. https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device/__init__.py#L61 and https://github.com/rm-hull/luma.lcd/blob/4575e4bdb07c9bb0859cd4b0d4c013175dd67fd8/luma/lcd/device.py#L235

thijstriemstra, 2021-01-13:
@rm-hull I was thinking about something like a `get_supported_modes()` (or sizes) method on `luma.core.device.device` that returns whatever dimensions the luma library supports.

For example, for `luma.lcd.device.st7735`, `get_supported_modes()` would currently return:

```python
[(160, 80), (160, 128), (128, 128)]
```

For `luma.oled.device.sh1106` it would have to include some extra data or we need to remap this dict action going on there:

```python
{
    (128, 128): dict(multiplex=0xFF, displayoffset=0x02),
    (128, 64): dict(multiplex=0x3F, displayoffset=0x00),
    (128, 32): dict(multiplex=0x20, displayoffset=0x0F)
}
```

Like the previous example, it should return a list:

```python
[(128, 128), (128, 64), (128, 32)]
```

but with some more metadata, so not a tuple but perhaps a dict with width/height keys?

Thoughts?

rm-hull, 2021-01-13:
What or how would you make use of `get_supported_modes()`?

I think in principle, yes we could readily add that to every device class (possibly as a class method?) but I’m wondering how a program could actually make use of it, because mostly if you have a display you almost certainly know up-front what the size is.

thijstriemstra, 2021-01-13:
I would use it in the luma.examples, and show what sizes are supported in the error message. And yea overall it's cleaner to have it in a central place, I often check the source code what sizes are supported and having it in a general place across all luma devices would be nice.

It's mostly a refactor thing to make things uniform.