rm-hull/luma.core #164: Cannot import any luma project

thijstriemstra, 2019-06-15:
I think this is related to #146 

when i checkout luma.core, `pip install -e .` it and then go to luma.led_matrix, I am unable to import luma.led_matrix.

```python
pi@bar:~/projects/pi-test/luma.led_matrix $ python
Python 3.7.3 (default, Jun 15 2019, 18:14:45) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import luma.core
>>> import luma.led_matrix
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'luma.led_matrix'
>>> 
```

thijstriemstra, 2019-06-15:
@rm-hull are you able to reproduce this?

rm-hull, 2019-06-15:
I don't tend to ever use `pip install -e .`
Only ever install version from pypi.
These days I generally use pipenv to install a virtualenv with just the dependencies I need.

What does your /use/local/lib/python3.7/dist-packages/Lima tree look like? (Or is that site-packages, can never remember)

thijstriemstra, 2019-06-15:
I also use a virtualenv. when using `pip install -e .` it creates `/home/pi/.virtualenvs/foo/lib/python3.7/site-packages/luma.core.egg-link`.

thijstriemstra, 2019-06-15:
@rm-hull can you try with the latest setuptools? they made some changes related to namespace packages, e.g. https://github.com/pypa/setuptools/pull/1402 and see https://github.com/pypa/setuptools/blob/master/CHANGES.rst#v4010

looks like we should use the new `find_namespace_packages()`?

thijstriemstra, 2019-06-15:
when i revert the changes made in PR #146, re-install luma.core with that change, and then try to install luma.led_matrix, it throws:

```
error: Namespace package problem: luma is a namespace package, but its
    __init__.py does not call declare_namespace()! Please fix it.
    (See the setuptools manual under "Namespace Packages" for details.)
    "
```

so the change in #146 seems a correct one.

thijstriemstra, 2019-06-15:
I'm now looking at the recommendations described on https://packaging.python.org/guides/packaging-namespace-packages and hopefully will be able to fix this.

thijstriemstra, 2019-06-15:
Going to test this approach across the sub-projects: https://packaging.python.org/guides/packaging-namespace-packages/#pkgutil-style-namespace-packages

Every `luma/__init__.py` needs to contain:

```python
__path__ = __import__('pkgutil').extend_path(__path__, __name__)
```

thijstriemstra, 2019-06-15:
cc @Gadgetoid

thijstriemstra, 2019-06-15:
I cannot reproduce this on Python 2.7.13, only on Python 3.5.3 and Python 3.7.3 (have not tested Python 3.6 or 3.4).

thijstriemstra, 2019-06-15:
Finally fixed this, sigh. Basically every project (inc core) needs to have this in `luma/__init__.py`:

```python
try:
    __import__('pkg_resources').declare_namespace(__name__)
except ImportError:
    __path__ = __import__('pkgutil').extend_path(__path__, __name__)
```

and use `find_packages()` and `namespace_packages` in `setup.py`:

```python
packages=find_packages(),
namespace_packages=["luma"],
```

PRs will follow.

closed 2019-06-15 by rm-hull

thijstriemstra, 2019-06-15:
all projects now use the same imports and setup.py metadata. can you make a new release for each project @rm-hull? :cactus:

rm-hull, 2019-06-16:
Will do, but it will be later on this evening

rm-hull, 2019-06-16:
Think I've published all sub-projects with your fix.Give me a nudge if I've missed anything.

Awesome effort, as ever!   :beers:

PR #165 (2019-06-15): namespace fix
fixes #164