rm-hull/luma.core #169: Setup on sponsor button for each project

thijstriemstra, 2019-07-25:
"Sponsorships help your community know how to financially support this repository."

See settings of Github repository.

thijstriemstra, 2019-07-25:
Looks like https://opencollective.com might be a good one to handle this?

thijstriemstra, 2019-12-21:
@rm-hull thoughts on this?

rm-hull, 2019-12-22:
Not really sure about how I feel about this, more out of ignorance than anything else, but some initial thoughts/questions are:
* if someone sponsors a new feature (for example) is there an obligation for that feature to be completed?
* what should the funds collected through sponsorship be spent on? New devices? Given to people who implement requested features? Given to people who actively help answering newbie questions?
* do funds collected need to be accounted for and declared on tax returns?
* for luma projects specifically, is there enough demand for new features or devices to support sponsorship (i.e. is it worth the effort)?
* who is going to administer such a sponsorship scheme?
* should we create one big mono-repo with all sub-projects subsumed first?

Given that, happy to trial it for something like 6 months