rm-hull/luma.core #170: Support arbitrary GPIO pins for SPI chip select

ric2b, 2019-11-29:
The goal would be to support more than 2 devices.

I don't mind implementing this and submitting a PR, I'm just wondering if this is something you'd accept or if you think the library should be limited to what spidev supports.

The idea would be to initialize spidev with [`no_cs`](https://github.com/doceme/py-spidev#settings) and control the chip select with GPIO from luma itself.

ric2b, 2019-12-06:
By the way, if anyone reads this and is interested in this functionality you can just wrap luma's spi class like this:

```python3
from luma.core.interface.serial import spi
from luma.lcd.device import ili9341
class CustomChipSelectSPI(spi):
    def __init__(self, *args, gpio_CS=None, **kargs):
        self.gpio_CS = gpio_CS
        if gpio_CS:
            GPIO.setup(gpio_CS, GPIO.OUT)
            GPIO.output(gpio_CS, False)
        super()._init__(*args, **kargs)
        if gpio_CS:
            GPIO.output(gpio_CS, True)

    def _write_bytes(self, *args, **kargs):
        if self.gpio_CS:
            GPIO.output(self.gpio_CS, False)
        super()._init__(*args, **kargs)
        if self.gpio_CS:
            GPIO.output(self.gpio_CS, True)

device = ili9341(CustomChipSelectSPI(device=0, port=0, gpio_CS=23)
```

Then you can instantiate multiple `CustomChipSelectSPI`'s, if can set them all to the same `device` and `port`.

rm-hull, 2019-12-06:
Apologies for not responding sooner, I’ve been tremendously distracted with work and home life lately. I’m hoping things will calm down come the new year.

What you propose looks interesting and clearly works. 

I thought the RPI had two SPI buses and each has two ports (not sure if I’ve got the correct nomenclature here), so that would give you four addressable devices. 

 Happy to incorporate this capability in luma.core (and we should also think about providing a cmd-line arg in luma.examples to allow it to be specified when running the demos)... and I’ll try and be a bit less sketching in responding 😬

ric2b, 2019-12-09:
Hey, no problem, I totally understand and I'm not in a hurry :)

Yes, the goal here is to handle more than 4 devices on a single bus. Even for 4 devices, wiring 2 separate buses doesn't sound great.

I already tested the above implementation with 3 devices, it works as expected but you need to avoid using the native CS pin since that will activate for all byte transfers (haven't tried the `no_cs` option on spidev yet but that might be enough to fix it).

I can prepare a PR if you like the idea, let me know if there's anything you want me to take into account or think about.

PR #172 (2020-01-12): Add support for using any GPIO pin as a CS pin
fixes #170

There seems to be a bug with spidev where it ignores the `no_cs` option: https://github.com/doceme/py-spidev/issues/46

Besides that, I've tested it locally with 3 screens on a single `(port, device)` and it works as expected.
closed 2020-01-12 by rm-hull