rm-hull/luma.core #178: suggestion:  move examples.demo_opts.get_device() into core

BeauSlim, 2020-07-16:
I've been messing around with a library I'm writing and trying it with numerous displays, including an ssd1306, an ili9341 and an st7789 in addition to the pygame emulator.  A config file (same file name, different contents) helps when testing on different displays, so I have used examples' get_device() as a start.  

While it isn't difficult to do 

from luma.core import cmdline
parser = cmdline.create_parser(description='luma config')
config = cmdline.load_config('device.conf')
args = parser.parse_args(config)
device = cmdline.create_device(args)

It occurs to me that something like device=conf2dev(dev_conf='device.conf') might be useful for users in general and really should be built into luma.core.  Probably just the file-based config not the command line stuff, since that might be messy.  

Or maybe the use of config files should just be documented better?  

Either way I'm happy to work it up as a pull request.

thijstriemstra, 2020-07-17:
> Either way I'm happy to work it up as a pull request.

Sounds good!

rm-hull, 2020-07-21:
Would be good, yes please.

Something like `device = create_device_from(filename='ssd1306.conf')` seems self-descriptive.
Also the use of config files should definitely be better documented too but only if you have time

BeauSlim, 2020-07-22:
Cool.  Thanks for guidance on the preferred naming.  I'll give the docs a try as well.  
Something has come up on my end so it might be a couple weeks.

thijstriemstra, 2020-10-20:
ping @BeauSlim

BeauSlim, 2021-01-15:
Sorry for leaving this hanging.  My laptop died and I could no longer code all comfy on my couch.   I have a shiny new M1 Air now, this is still on my list and I should have some time soon.