rm-hull/luma.core #185: Autostarting a python3 script at boot not working in /etc/rc.local and systemd

suptronics, 2020-09-05:
Hi Friends,

Help me! please.
I am having an error running my script on boot (2020-08-20-raspios-buster).
I have tested my script manually and it runs without errors, I also checked that it imports everything that is needed. But when I run it on boot (tried rc.local and systemd but nothing happens) it seems it can't import modules below. Is there anything else that I needed to do? Many thanks.

pi@raspberrypi:~ $ sudo bash -c 'python /home/pi/test.py > /home/pi/test.log 2>&1' &
pi@raspberrypi:~ $ Traceback (most recent call last):
  File "/home/pi/test.py", line 10, in <module>
    from luma.core.interface.serial import spi, noop
ImportError: No module named luma.core.interface.serial

My script
--------------------------------------------------------------------------------------------------------------
#!/usr/bin/python3

import RPi.GPIO as GPIO
from time import sleep, strftime
from datetime import datetime

from luma.core.interface.serial import spi, noop
from luma.core.render import canvas
from luma.core.virtual import viewport
from luma.led_matrix.device import max7219
from luma.core.legacy import text, show_message
from luma.core.legacy.font import proportional, CP437_FONT, LCD_FONT

serial = spi(port=0, device=0, gpio=noop())
device = max7219(serial, width=32, height=16, block_orientation=90)
device.contrast(5)
virtual = viewport(device, width=32, height=16)
show_message(device, 'Welcome', fill="white", font=proportional(LCD_FONT), scroll_delay=0.03)

try:
    while True:
        with canvas(virtual) as draw:
            text(draw, (2, 0), "hello", fill="white", font=proportional(LCD_FONT))
            text(draw, (2, 9), datetime.now().strftime('%I:%M'), fill="white", font=proportional(LCD_FONT))

except KeyboardInterrupt:
    GPIO.cleanup()

thijstriemstra, 2020-09-05:
> tried rc.local and systemd but nothing happens)

What does your systemd file look like?

rm-hull, 2020-09-05:
Try running in rc.local/systems using sudo with -u option: ` sudo -u pi bash -c 'python /home/pi/test.py > /home/pi/test.log 2>&1' & ` ?

suptronics, 2020-09-05:
> Try running in rc.local/systems using sudo with -u option: `sudo -u pi bash -c 'python /home/pi/test.py > /home/pi/test.log 2>&1' &` ?

Thanks. No, that command is for debugging.  I modified the rc.local using "sudo nano /etc/rc.local" and put "sudo python3 /home/pi/test.py &" before the exit 0.

suptronics, 2020-09-05:
@thijstriemstra 
Thanks for your time.  

1) Create the .service file in the systemd directory using  sudo nano /lib/systemd/system/test.service
2) Enter the following text into the document:
-----------------------------------
[Unit]
Description=test LED matrix
After=multi-user.target

[Service]
ExecStart=/usr/bin/python3 /home/pi/test.py

[Install]
WantedBy=multi-user.target
----------------------------------
3) Tell systemd to recognize the service using  sudo systemctl daemon-reload
4) Tell systemd to start on boot using  sudo systemctl enable test.service
5) Reboot the system but nothing happens.

suptronics, 2020-09-05:
@rm-hull @thijstriemstra 
Updated. after running a full system update, it can finally run on startup. so seems 'Updating and upgrading the OS are important for Autostarting . Thanks.

closed 2020-09-05 by suptronics