rm-hull/luma.core #204: Use time.monotonic_ns

thijstriemstra, 2020-10-20:
Python 3.7 introduces `time.monotonic_ns`. [PEP564](https://www.python.org/dev/peps/pep-0564/) explains the rationale but what stood out to me is:

> A server is running for longer than 104 days. A clock is read before and after running a function to measure its performance to detect performance issues at runtime. Such benchmark only loses precision because of the float type used by clocks, not because of the clock resolution.

> On Python microbenchmarks, it is common to see function calls taking less than 100 ns. A difference of a few nanoseconds might become significant.

More precision is always better.

thijstriemstra, 2020-10-24:
thoughts @rm-hull? It could be introduced in 2.0 as well.

rm-hull, 2020-10-24:
Yep, we can add this; was there anywhere specifically it would be useful - presumably updating existing instances of `time.monotic()` only?

thijstriemstra, 2020-10-24:
yea my thoughts as well. I'll start a branch.

thijstriemstra, 2021-01-13:
This can be easily supported once Python 3.6 reaches EOL.

thijstriemstra, 2022-05-07:
> This can be easily supported once Python 3.6 reaches EOL.

looks like this happened on [23 Dec 2021](https://endoflife.date/python)

closed 2022-10-16 by rm-hull

thijstriemstra, 2022-10-29:
more good `time.sleep()` improvements in python 3.11: https://docs.python.org/3.11/whatsnew/3.11.html#time

commit 76326e86 (2022-10-16): Drop support for Python 3.6 (#255)

* drop support for Python 3.6 (fixes #252)

* use time.perf_counter_ns (fixes #204)

* remove cmdline.make_serial, deprecated since 1.17.1 (fixes #250)

* ci: stop testing against python 3.6

* remove whitespace

* fix pillow deprecation warning

* fix Pillow deprecation warning

* tests: update image

* remove unused library

* require pillow 9.2.0 or newer

* update changelog

* use f-string

* use implicit namespace package (fixes #254)

* remove deprecated option