rm-hull/luma.core #213: Multiple spi displays -> need help

Maschine2501, 2020-11-03:
Hi!
I want to use more than one display on one spi line.
I researched this, and came to the conclusion, that only the cs/ce pin must be different.
So I wired two ssd1351 displays up and connected them as followed (BCM Numbers)

I guessed that I need to initialize the seccond display with [gpio_cs_spi](https://github.com/rm-hull/luma.core/blob/bc788fd0161884765ddef60100030a474866a300/luma/core/interface/serial.py#L331)...

#### 1st Display:
Mosi -> 10
SCLK -> 11
RES ->    25
DC ->     24
CS ->      8

#### 2nd Display:
Mosi -> 10
SCLK -> 11
RES ->    25
DC ->     24
CS ->      7

#### Results:
Only on Display 2 (cs - 7) i get a picture.
If i wire it up so that dc, reset and cs are individual for each display and only Mosi and Sclk are shared, i get the same results.

#### Type of Raspberry Pi
Raspberry Pi 4 / 4GB
spi enabled
latest buster lite

#### Linux Kernel version
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

#### Expected behaviour
Both displays will display a picture (in the demo code below it is a text...) undependend to each other.
Goal is that i can display something else on each display with minimal use of other GPIO's.

#### Actual behaviour
Only one Display shows something.
Allready testet the displays, both work just fine in "mono" mode without the other.
I even tried an ssd1322 + ssd1351 same result.

#### The Code: 
( I've left the stuff I tried in the code, but commented)
```
#!/usr/bin/python3
import os
import sys
import time
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont
from luma.core.interface.serial import spi
#from luma.oled.device import ssd1322
from luma.core.interface.serial import gpio_cs_spi
from luma.oled.device import ssd1351

width1322 = 128
height1322 = 128
width1351 = 128
height1351 = 128

font = ImageFont.load_default()

interface22 = gpio_cs_spi(device=0, port=0, gpio_DC=14, gpio_RST=15, gpio_CS=7)
#interface22 = gpio_cs_spi(device=0, port=0, gpio_CS=8)
#interface22 = spi(device=0, port=0)
#oled22 = ssd1322(interface22) 
oled22 = ssd1351(interface22) 
oled22.clear()
interface51 = gpio_cs_spi(device=1, port=0, gpio_CS=7)
oled51 = ssd1351(interface51) 
oled51.clear()

image22 = Image.new('RGB', (width1322, height1322))
image51 = Image.new('RGB', (width1351, height1351))

draw22 = ImageDraw.Draw(image22)
draw51 = ImageDraw.Draw(image51)

def ssd1322picture():
    draw22.text((0, 20), "SSD1322", font=font, fill='white')
    oled22.display(image22)

def ssd1351picture():
    draw51.text((0, 64), "SSD1351", font=font, fill=(0, 255, 0))
    oled51.display(image51)

while True:
    ssd1322picture()
    ssd1351picture()
    time.sleep(10.0)
    oled22.clear()
    oled51.clear()
```

#### Can anyone help me with this topic? Is it even possible?

CRImier, 2020-11-03:
quick answer - you probably need separate RESET pins.

Maschine2501, 2020-11-03:
@CRImier Thank you for your quick reply!
I allready tried that...
```
interface22 = gpio_cs_spi(device=0, port=0, gpio_DC=14, gpio_RST=15, gpio_CS=7)
```
...and of course: i changed the wirimg regarding this... same results...
:-(

CRImier, 2020-11-03:
you cannot parallel CS and RST, you can likely parallel CS - and, of course, SCK and MOSI. Not noticing any other problems, except that I'd try and exclude factors. Connect a LED from ground/VCC to every one of these lines (resistor optional, 1K-10K range), is there any activity on every one of these lines when you run the sketch.

CRImier, 2020-11-03:
Currently, you have:
```
init interface22
init display22
init interface51
init display51
update display22
update display55
```

What if you change order of operations? Say, you init both interfaces first and then init both displays, instead of interleaving interface inits. Since only one display works, which things do you need to swap places that'd result in only the first display working? That operation is probably what's causing the displays to not work.

Also:
```python
while True:
    ssd1322picture()
    ssd1351picture()
    time.sleep(10.0)
    oled22.clear()
    oled51.clear()
```
the `clear()` functions will execute right before `*picture()` functions will execute again with the loop's new iteration - it might be that you'll never actually see the displays clear, since the time between `clear()` and `*picture()` is so short. For testing data lines with the LED, I'd just do:
```python
while True:
    ssd1322picture()
    ssd1351picture()
```
just so that the data lines switch all the time and it's easier for you to notice which one's not working (except RST, RST will only switch when you run the sketch and that's expected).

CRImier, 2020-11-03:
Oh, also. You can parallel RST, but you need to ensure that the first display to be init is the one that receives the rst= pin, second has no rst= pin supplied. As you only need to toggle the RST pin once before sending display initialization commands (that's what luma code does), you'll effectively do "toggle both resets at once, then send init commands to display1 and send init commands to display2" with such a configuration. As long as the second display to be init doesn't result in the RST pin being toggled again, it should work just fine (otherwise, the first display would be reset back into its unconfigured state).

Maschine2501, 2020-11-04:
@CRImier : Thank you! Your suggestions made it!!!
First Display now is init witth a dedicated RST, seccond got a dedicated cs... Both working fine now!
Here is the code that works:
```
#!/usr/bin/python3
import os
import sys
import time
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont
from luma.core.interface.serial import spi
#from luma.oled.device import ssd1322
from luma.core.interface.serial import gpio_cs_spi
from luma.oled.device import ssd1351

width1322 = 128
height1322 = 128
width1351 = 128
height1351 = 128

font = ImageFont.load_default()

interface22 = spi(device=0, port=0, gpio_RST=14)
interface51 = gpio_cs_spi(device=1, port=0, gpio_CS=7)
oled22 = ssd1351(interface22) 
oled51 = ssd1351(interface51) 
oled22.clear()
oled51.clear()

image22 = Image.new('RGB', (width1322, height1322))
image51 = Image.new('RGB', (width1351, height1351))

draw22 = ImageDraw.Draw(image22)
draw51 = ImageDraw.Draw(image51)

def ssd1322picture():
    draw22.text((0, 20), "SSD1322", font=font, fill='white')
    oled22.display(image22)

def ssd1351picture():
    draw51.text((0, 64), "SSD1351", font=font, fill=(0, 255, 0))
    oled51.display(image51)

while True:
    ssd1322picture()
    ssd1351picture()

```

CRImier, 2020-11-04:
Happy to hear that! Don't forget to close this issue ;-P

thijstriemstra, 2020-11-04:
This ticket should be moved to luma.oled I think @rm-hull

Maschine2501, 2020-11-04:
Thank you for assistance, maybe this topic shouöd be inclouded in the docs/wiki?

closed 2020-11-04 by Maschine2501

thijstriemstra, 2020-11-04:
@Maschine2501 sounds good, feel free to open a pull request for that in the luma.oled project.

CRImier, 2020-11-04:
yeah, it would be very valuable for other people if you could document the important points (what was it that made a difference for you) and add a bit of example code.