rm-hull/luma.core #216: Reduce the build jobs on Travis ... or use Github Actions

rm-hull, 2020-11-20:
Email today from travis:

> Welcome to Travis CI, Richard Hull!You are on Free.You have 10,000 credits left - these will begin counting down automatically as soon as you run your first build.You can use your credits to build on both private and open-source repositories using Linux, macOS, and Windows OS.Use the `Sign up now` button below to sign up for a bigger plan to ensure the builds for your private repositories continue uninterrupted.If you're building open-source public repositories and want to signup for an open-source plan, please contact Travis CI.
--
You have 10,000 Credits left for building on Travis CI!

Now that travis is no longer 'free', and since every time a build across any luma.* project is triggered, it will kick off multiple jobs to run the same tests on different versions of python.

This will quickly drain down the 10000 credits, so we have a choice:

1. Remove the build jobs for different versions of python (this reduces the rate at which the credits will go down, but not ideal in the longer term)
1. Contact travis and appeal for an open-source plan (this may be successful, but who knows?)
1. Move all CI operations to github actions instead (free right now, and some small effort required to convert current CI)

Any thoughts on preferred next step @thijstriemstra  ?

thijstriemstra, 2020-11-20:
> Contact travis and appeal for an open-source plan 

Let's try this. And otherwise 3? I got the same email, thanks for the headsup.

dhrone, 2020-11-23:
I'm all for seeing if Travis can accommodate and if the reasoning they've given for the change is real, perhaps they will.  That said, the move to actions doesn't look that painful...

thijstriemstra, 2020-11-23:
I moved some of my private repos over to github actions and it seems to be really fast. Unfortunately it also has limits..

rm-hull, 2020-11-23:
I fired off a request to Travis support and they initially came back quite quickly with a canned response, but nothing substantive since I responded.

rm-hull, 2020-11-26:
Still no response from travis support.
I just ran a build for luma.core and it appears to have cost 730 credits (total free credits are 10000), so by my reckoning, that will be down to zero after another 13 or 14 runs ... 

:-s

thijstriemstra, 2020-11-27:
Ok I will make a github actions pr.

PR #218 (2020-11-29): ci: switch to github actions
fixes #216
closed 2020-11-29 by rm-hull