rm-hull/luma.core #217: Possible to use diff-algorithm with linux_framebuffer device?

mattblovell, 2020-11-27:
#### Type of Raspberry Pi

Raspberry Pi 3 Model B (v1.2)

#### Linux Kernel version

`Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux`
(Buster release of Raspberry Pi OS)

#### Question

Thanks for adding the linux_framebuffer device into luma.core!  I'm just getting the chance to experiment with it.

Is it possible at all to make use of luma's **diff_to_previous** algorithm for updates to the fbdev framebuffer?  I ask just since the full update (I assume that's what is currently being used) is taking a little over 1/4 of a second.

With an update loop that looks as follows:
```
        while True:
            try:
                start_time = datetime.now()
                draw.rectangle([(0,0), (frame_size[0],frame_size[1])], 'black', 'black')
                draw.text(( 5, 5), "Hello, world",  fill='white', font=font_main)
                device.display(image)
                #update_display()   -- isolate update time to just fb update
                print(datetime.now(), "update_display() took ", datetime.now() - start_time)
```

some representative results are
```
2020-11-27 11:27:25.556874 update_display() took  0:00:00.292568
2020-11-27 11:27:26.745018 update_display() took  0:00:00.276966
2020-11-27 11:27:27.929180 update_display() took  0:00:00.273018
2020-11-27 11:27:29.115797 update_display() took  0:00:00.275477
2020-11-27 11:27:30.320051 update_display() took  0:00:00.293113
2020-11-27 11:27:31.510489 update_display() took  0:00:00.279289
2020-11-27 11:27:32.691867 update_display() took  0:00:00.270262
2020-11-27 11:27:33.900963 update_display() took  0:00:00.297949
2020-11-27 11:27:35.108634 update_display() took  0:00:00.296486
2020-11-27 11:27:36.315688 update_display() took  0:00:00.295907
2020-11-27 11:27:37.499120 update_display() took  0:00:00.272288
2020-11-27 11:27:38.698876 update_display() took  0:00:00.288624
2020-11-27 11:27:39.891200 update_display() took  0:00:00.281173
2020-11-27 11:27:41.077159 update_display() took  0:00:00.274793
```

The above is with an ILI9341-based display driven using tinyDRM's `mi0283qt` driver.  The dtoverlay file is currently specifying a SPI frequency of 32 MHz.  The SPI bus could certainly be set faster, but I thought I'd inquire about the fbdev update.

Thanks again!
  Matt

rm-hull, 2020-11-28:
On a model 4 RPi, I saw 14 FPS (see https://github.com/rm-hull/luma.core/pull/194#issuecomment-713699572) against an ILI9341. I was using a different kernel driver though, so it is curious to see yours with a rate of approx 4 frames/second.

What is returned if you did `cat /sys/class/graphics/fb0/bits_per_pixel` ?

It may well be possible to use the diff-to-previous algorithm, but I’d have to look more into the Linux framebuffer documentation to see if it is possible to write to specific areas of the framebuffer display area

mattblovell, 2020-11-29:
> What is returned if you did `cat /sys/class/graphics/fb0/bits_per_pixel` ?

Changing the above to fb1, the result is **16**.  (fb0, for the RPi3's HDMI display, returns 32.)

The fb_sierpienski demonstration from [fbmark](https://github.com/caramelli/fbmark) executes with the following times, using the `mi0283qt` overlay driver I mention:

```
Sierpinski frame buffer test bench

Benchmarking     1024 iterations:  2879.80 Frames/second
Benchmarking     2048 iterations:  1672.48 Frames/second
Benchmarking     4096 iterations:   877.95 Frames/second
Benchmarking     8192 iterations:   454.31 Frames/second
Benchmarking    16384 iterations:   229.33 Frames/second
Benchmarking    32768 iterations:   115.90 Frames/second
Benchmarking    65536 iterations:    58.74 Frames/second
Benchmarking   131072 iterations:    29.47 Frames/second
Benchmarking   262144 iterations:    14.70 Frames/second
Benchmarking   524288 iterations:     7.44 Frames/second
```

It's measuring a mix of CPU calculation and frame rate, though.

The fb_rectangle demo gets this result:

`Benchmarking 80x60 size: 41.41 MPixels/second`

> It may well be possible to use the diff-to-previous algorithm, but I’d have to look more into the Linux framebuffer documentation to see if it is possible to write to specific areas of the framebuffer display area

Page 5 of the slides from this presentation suggest it should be:

**FOSDEM 2020: Back to the Linux Framebuffer!**
https://archive.fosdem.org/2020/schedule/event/fbdev/

I don't remember if I shared those slides with you earlier; I found them quite interesting!  This (now old, just because of fbdev being deprecated) kernel documentation also suggests seeking is possible:

https://www.kernel.org/doc/Documentation/fb/framebuffer.txt

> ```
> 2. Programmer's View of /dev/fb*
> --------------------------------
> 
> As you already know, a frame buffer device is a memory device like /dev/mem and
> it has the same features. You can read it, write it, seek to some location in
> it and mmap() it (the main usage). The difference is just that the memory that
> appears in the special file is not the whole memory, but the frame buffer of
> some video hardware.
> 
> /dev/fb* also allows several ioctls on it, by which lots of information about
> the hardware can be queried and set. The color map handling works via ioctls,
> too. Look into <linux/fb.h> for more information on what ioctls exist and on
> which data structures they work. Here's just a brief overview:
> ```

Alternatively, does `linux_framebuffer` hold the fd for the duration of execution or open and close it as needed?  

Thanks,
  Matt

rm-hull, 2020-11-29:
Ok, so the code snippet on page 5 of that slide deck is all we need to make the diff-to-previous algorithm work. I will have a go at it tonight!

rm-hull, 2020-11-29:
@mattblovell how did you get the mi0283qt driver working? I am finding the tinydrm documentation impenetrable to say least.

mattblovell, 2020-11-29:
> how did you get the mi0283qt driver working?

I modified the overlay source file (.dts) from the tinyDRM installation to match the wiring I was using with luma.lcd and comment out the touchscreen driver.  You can find the modified file in my kodi_panel repo:

https://github.com/mattblovell/kodi_panel/blob/master/other/tinydrm-rpi-display.dts

Comments at the start of that file provide the `dtc` incantation by which one obtains the binary overlay file, as well as the run-time `dtoverlay` command to make use of it.  Since I wanted to keep switching back and forth between that driver and having luma.lcd drive the display itself via SPI, I never modified my `config.txt` file.

Hope that helps!

rm-hull, 2020-11-29:
Alright, I'll give that a try. In the meantime, I just re-enabled the pitft28-resistive overlay as i had previously documented.

The good news is that it is possible to use the diff-to-previous algorithm with the linux framebuffer - see PR #219 - If you want to try it, checkout that branch and install with `pip install -e .` from the luma.core directory.

The tests will need repairing, but it shouldn't be too long before it is released and published to pypi. 

Many thanks for the inspiration and doc links for this!

mattblovell, 2020-11-29:
That's great!  Hopefully I'll get a chance later today to try it.

Thanks!

mattblovell, 2020-11-30:
Results of trying the update loop I posted above with the feature/diff-to-previous-linux-framebuffer branch:

```
2020-11-29 17:32:12.762971 update_display() took  0:00:00.015327
2020-11-29 17:32:13.689408 update_display() took  0:00:00.015192
2020-11-29 17:32:14.615960 update_display() took  0:00:00.015292
2020-11-29 17:32:15.542564 update_display() took  0:00:00.015290
2020-11-29 17:32:16.469719 update_display() took  0:00:00.015899
2020-11-29 17:32:17.396221 update_display() took  0:00:00.015266
2020-11-29 17:32:18.322851 update_display() took  0:00:00.015387
```

That loop doesn't actually update anything, but now it could do so at >60 fps.  So, pretty good!

I see that a review of the PR was requested.  I'll go see what I need to do...

PR #219 (2020-11-30): Linux framebuffer to use diff-to-previous algorithm
Doing a full-frame render, I observe 14 frames/sec on RPi 4 using the bounce.py demo on the linux framebuffer pseudo-device.

With this PR, where it uses the diff-to-previous render algorithm, the same bounce demo shows the frame rate at approx 30 frames/sec

Fixes #217
closed 2020-11-30 by rm-hull