rm-hull/luma.core #222: Supporting 32-bit depth for linux_framebuffer

mattblovell, 2020-12-06:
#### Type of Raspberry Pi

Raspberry Pi Zero W

#### Linux Kernel version

Linux raspberrypi 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l GNU/Linux

(Dec 2020 update of buster for Raspberry Pi OS)

#### Enhancement question

I'm trying out a small HDMI-attached display that I had working previously under an RPi 4.  For some reason, on the Zero, the framebuffer device gets created with 32-bit depth.  As a consequence, luma.core doesn't want to start.  Invoking one of the examples like

`python3 demo.py --display linux_framebuffer --framebuffer-device /dev/fb0`

yields

```
  File "/home/pi/projects/luma.core/luma/core/device.py", line 212, in __init__
    assert self.bits_per_pixel in image_converters, f"Unsupported bit-depth: {self.bits_per_pixel}"
AssertionError: Unsupported bit-depth: 32
```

Unlike the RPi4, I think the Zero continues to use legacy driver for its display.  With that, one can specify

`framebuffer_depth=24`

in `/boot/config.txt`.  Using that, and rebooting, I am able to get the examples to start, but there is an odd color-shift...

- red appears blue
- blue is red
- orange is light blue
- yellow is sky blue
- green is still green

Adding `--bfr` to the command line doesn't change the results.

rm-hull, 2020-12-07:
Couple of things to unpack here:
* I didn’t have a 32 bit depth device to test with so chose not to provide an implementation (it did not occur to me at the time to plug in a HDMI display, which seems obvious now)
* It does sound like flag needs to be supplied to flip red & blue for 24 bit displays. I think we just assumed pixels would be RGB, rather than BGR. Supporting BGR may well incur a performance penalty though in that mode.

mattblovell, 2020-12-07:
> * It does sound like flag needs to be supplied to flip red & blue for 24 bit displays. I think we just assumed pixels would be RGB, rather than BGR.

I was surprised as well, since I'm pretty sure I tried out that HDMI display on a Pi4 with luma.lcd and linux_framebuffer without issue.  (That was the setup that triggered my question about the diff algorithm for fbdev.)  I may not have been using the legacy driver with that setup, though.

Perhaps it's an artifact of using `config.txt` to limit color depth.  The Raspberry Pi page

https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

documenting `framebuffer_depth` helpfully notes that it "May result in a corrupted display". 

Maybe it will be better to pump out an unused alpha channel than to deal with the color swapping!

commit 5bd0bd26 (2020-12-11): linux framebuffer: Support BGR, RGBA & BGRA colorspace displays (#223)

Fixes #222
closed 2020-12-11 by rm-hull