rm-hull/luma.core #224: Use github actions to publish to pypi

thijstriemstra, 2020-12-12:
See https://help.github.com/en/actions/language-and-framework-guides/using-python-with-github-actions#publishing-to-package-registries

thijstriemstra, 2021-01-13:
@rm-hull I see you currently use this script: https://github.com/rm-hull/luma.core/blob/master/bin/make_release.sh

But basically now github actions would publish to pypi (using that command) whenever a new tag is created. This way it's as simple as creating a new tag and releasing can be done remotely.

Thoughts?

rm-hull, 2021-01-13:
Great idea. Presumably all we need (aside from the action yaml) is the pypi creds stored as a secret in GitHub?

thijstriemstra, 2021-01-16:
Yea follow [these instructions](https://docs.github.com/en/actions/reference/encrypted-secrets#creating-encrypted-secrets-for-an-environment) to create 2 secrets:

- `PYPI_USERNAME`
- `PYPI_PASSWORD`

Please let me know when you added these @rm-hull

thijstriemstra, 2021-01-16:
@rm-hull actually scratch previous comment, this approach uses pypi API key which seems a better solution to me: https://github.com/marketplace/actions/pypi-publish

More info on pypi api: https://pypi.org/help/#apitoken

rm-hull, 2021-01-22:
@thijstriemstra created

[image]

PR #232 (2021-01-28): Use github actions to publish to pypi
fixes #224 

depends on #233
closed 2021-03-08 by thijstriemstra