rm-hull/luma.core #235: IndexError: Font does not have ASCII code: 8211 (or 8212,8217,8220,8221)

dliloch, 2021-01-30:
linux software environment:
>uname -a
Linux MATRIX 5.4.83+ #1379 Mon Dec 14 13:06:05 GMT 2020 armv6l GNU/Linux

hardware:
Raspberry pi zero v2. driving 2 ALAMSCN MAX7219 Doxt Matrix Module 32x8 4 in 1 LED Display Module.

I call the bing news api via rapidapi and then try and display the headline upon the led matrix
The text sometimes has one or more of these 5 character ascii codes:
8211
8212
8217
8220
8221
I then get the following exception.

exception in GetSomeText:  2021-01-29 07:37:03  skipnws   7
^CTraceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/font.py", line 45, in __getitem__
    bitmap = self.font[ascii_code]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "rpimax7219led9.py", line 140, in <module>
    show_message(device, text, fill="white", font=proportional(LCD_FONT), scroll_delay=0.06)
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/__init__.py", line 81, in show_message
    w, h = textsize(msg, font)
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/__init__.py", line 29, in textsize
    src = [c for ascii_code in txt for c in font[ord(ascii_code)]]
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/__init__.py", line 29, in <listcomp>
    src = [c for ascii_code in txt for c in font[ord(ascii_code)]]
  File "/usr/local/lib/python3.7/dist-packages/luma/core/legacy/font.py", line 52, in __getitem__
    raise IndexError(f"Font does not have ASCII code: {ascii_code}")
IndexError: Font does not have ASCII code: 8211

My work around is to run the following replace python code on the
string before passing it to the luma.core show_message routine:

zz=chr(8212)+chr(8217)+chr(8211)+chr(8220)+chr(8221)
st6 = str(nwsart[skipnws]).translate({ord(c):' ' for c in zz})

My suggestion would be to fix the translate table font.py in LCD to translate
these 5 characters to a space.

rm-hull, 2021-01-30:
The legacy fonts typically only handle ASCII characters. Those ordinals are Unicode (utf-8 most likely) for which there is no bitmap. There is already a wrapper class called `tolerant` which ignores unknown values, See https://luma-core.readthedocs.io/en/latest/legacy.html#luma.core.legacy.font.tolerant

rm-hull, 2021-01-30:
See https://github.com/rm-hull/luma.core/blob/master/tests/test_tolerant.py for example usage

closed 2021-01-30 by rm-hull