rm-hull/luma.core #239: Crash @ Raspi3 install with "syntax error" ... already tried with Python 3.8.4

jesko42, 2021-04-25:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.
Raspberry Pi 3 Model B Rev 1.2
I know it's Raspi3, but ... seems to me that it has nothing to do with it

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.
`sudo ./setup.py install`
Shall install the packet, but crashes with 
"SyntaxError"

Important: I thought it could be that the python version was too low (3.5) so I downloaded python 3.8.4 source and compiled/installed it. But not changes ... even not for me ...

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

[luma.core_InstallCrash.txt](https://github.com/rm-hull/luma.core/files/6373331/luma.core_InstallCrash.txt)
As attached file ...

thijstriemstra, 2021-04-26:
@jesko42 your crash logs shows python 3.4.2 which is no longer supported. Please try 3.6 or newer.

> But not changes ... even not for me ...

Can you show the install log for 3.8.4?

jesko42, 2021-04-26:
I've checked what version is called using 'python --version' and it prints out:
```
[Python3.8.4_install.txt](https://github.com/rm-hull/luma.core/files/6373557/Python3.8.4_install.txt)
```

jesko42, 2021-04-26:
Hmmm, your proposal works without errors ... wondering ... have to test further ...

thijstriemstra, 2021-04-26:
it's a path/operating system thing I always have a hard time understanding myself. Stick to /usr/local/bin/python3 for now I guess.

thijstriemstra, 2022-10-07:
> Hmmm, your proposal works without errors 

Closing.

closed 2022-10-07 by thijstriemstra