rm-hull/luma.core #240: Support for Orange Pi GPIO module

edwardjarchibald, 2021-04-29:
I found the luma modules and examples today and worked through a few small issues that I found when getting everything running on an Orange Pi Zero LTS.  I would like to contribute the changes back to this code base.

#### Type of Raspberry Pi
Orange Pi Zero LTS running Armbian

#### Linux Kernel version
Linux elephant.local 5.10.16-sunxi #trunk SMP Wed Feb 17 06:59:22 PST 2021 armv7l armv7l armv7l GNU/Linux

#### Expected behaviour
After figuring out the correct configuration, shown here, I tried to run the 3d_box.py example.  I expected to see the example running on my ssd1351 display.  

--interface=spi
--gpio=OPi.GPIO
--spi-port=1
--spi-device=0
--display=ssd1351
--width=128
--height=128

#### Actual behaviour
I got the following traceback:

Traceback (most recent call last):
  File "3d_box.py", line 129, in <module>
    device = get_device()
  File "/root/luma.examples/examples/demo_opts.py", line 61, in get_device
    device = cmdline.create_device(args)
  File "/usr/local/lib/python3.8/dist-packages/luma/core/cmdline.py", line 246, in create_device
    device = Device(serial_interface=interface(), **params)
  File "/usr/local/lib/python3.8/dist-packages/luma/core/cmdline.py", line 152, in spi
    GPIO = self.__init_alternative_GPIO()
  File "/usr/local/lib/python3.8/dist-packages/luma/core/cmdline.py", line 208, in __init_alternative_GPIO
    GPIO.setmode(GPIO.BCM)
ValueError: Please use setboard(board) before setmode()

#### Proposed change to cmdline.py
1) Add a command line option called gpio-board which takes a string for the board definition from the GPIO library. In my example, the board is defined, in OPi.GPIO as 'ZERO' and is initialized as the integer '1'  
2)  If an 'alternative gpio module is specified, then also look to see if gpio-board is set and , if so, call the 'setboard()' function with the argument supplied.

Here are diffs for the proposed change made to luma.core 2.3.1:

202,204d201
<             if hasattr(self.opts, 'gpio_board') and self.opts.gpio_board is not None:
<                 GPIO.setboard(getattr(GPIO, self.opts.gpio_board))
< 
328d324
<     gpio_group.add_argument('--gpio-board', type=str, default=None, help='Board in use for alternative RPI.GPIO compatible implementation (SPI interface only)')

#### Behavior after the proposed change
All of the examples run perfectly except for those that rely on Raspberry Pi video etc.  Love the examples!

thijstriemstra, 2021-04-30:
thanks for the details. can you make a pull request?

edwardjarchibald, 2021-05-06:
Sure, happy to do that. It will take a bit more time before I get to it
since I'll need to put together a test etc.

Shaggy013, 2021-06-04:
and dont forget the 

lib.py 
def __rpi_gpio__(self):

little strange OPi-GPIO on the same github sp why is it not in luma ?

thijstriemstra, 2022-10-07:
> Sure, happy to do that. It will take a bit more time before I get to it

Still interested to make a PR?