rm-hull/luma.core #268: Raspberry Pi 5 + SPI Display

brickbots, 2023-12-16:
Hello!

I'm working to port some working code to bookworm on a Raspberry Pi 5.  This code works great on bookworm/RPI4, but with the RPI5 it's not able to initialize the SPI interface from the luma.core.interface.serial module.

Perhaps this is due to the new hardware controller on the 5?  

#### Type of Raspberry Pi
Raspberry Pi 5

#### Linux Kernel version
Bookworm
Linux pifinder 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

#### Expected behaviour
I'm trying to initialize a waveshare 1.5" RGB OLED display based on the ssd1351 controller.  Communication is via SPI

```python
from luma.core.interface.serial import spi

# init display  (SPI hardware)
serial = spi(device=0, port=0)
```

#### Actual behaviour
Crashes with the error below

 serial = spi(device=0, port=0)
             ^^^^^^^^^^^^^^^^^^^^^
  File "...python/venv/lib/python3.11/site-packages/luma/core/interface/serial.py", line 302, in __init__
    bitbang.__init__(self, gpio, transfer_size, reset_hold_time, reset_release_time, DC=gpio_DC, RST=gpio_RST)
  File "...python/venv/lib/python3.11/site-packages/luma/core/interface/serial.py", line 190, in __init__
    self._DC = self._configure(kwargs.get("DC"))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...python/venv/lib/python3.11/site-packages/luma/core/interface/serial.py", line 203, in _configure
    self._gpio.setup(pin, self._gpio.OUT)
RuntimeError: Cannot determine SOC peripheral base address

thijstriemstra, 2023-12-16:
From https://forums.raspberrypi.com/viewtopic.php?t=361218:

> RPi.GPIO, wiringPI, and pigpio do not work on the Pi5 which has new hardware for the GPIO.

Yikes..

brickbots, 2023-12-16:
Thank you!  Yes I've tried and no it does not work... I've played a little bit with GPIOD and it's related python bindings, which seems to be the method that people are coalescing around.  It theoretically works for RPI 3/4/5 and is fairly quick.... but I've not seen anything about SPI or I2C comms using that framework, just raw GPIO on/off.  

I'll keep looking and if I can figure out the 'new' way to do SPI, maybe work on a PR.  Just mainly wanted to see if anyone else had jumped into this yet or if there was some more clearcut solution.

thijstriemstra, 2023-12-16:
I wasn't in a rush to get a new Pi5 so haven't run into the issue yet but surprised that no one posted a bug report before, also didn't see anything on twitter etc. Releasing a new Pi without RPI.GPIO etc software support is quite strange by rpi foundation.

brickbots, 2023-12-16:
Agreed... even if the underlying hardware and memory mapping has changed, you'd think there could be some wedge or abstraction to allow most (all?) of the existing GPIO code to work....

Thank you for taking the time to think about this, and for the incredible Luma project.  If it's okay with you, I'll keep this open as a reminder to keep checking back on the state of play for python/spi bindings.

outdoorbits, 2023-12-30:
I just tested [gpiozero](https://gpiozero.readthedocs.io/en/latest/) as an alternative to RPi.GPIO for my project [little-backup-box](https://github.com/outdoorbits/little-backup-box) for the Raspberry Pi 4 and 5. This works for me on both the old and new hardware. However, I only tested it with inputs (buttons), not with outputs.