rm-hull/luma.core #283: Dependency on RPi.GPIO

Pako2, 2024-07-23:
Dependency on RPi.GPIO is a problem because RPi.GPIO does not support RPi5 and has problems with recent kernels. The temporary solution is of course to install [lgpio](https://abyz.me.uk/lg/py_lgpio.html) and [rpi-lgpio](https://github.com/waveform80/rpi-lgpio) (which emulates RPi.GPIO). It would be nice if the dependency on [rpi-lgpio](https://github.com/waveform80/rpi-lgpio) was removed and luma.core used lgpio directly.

thijstriemstra, 2024-07-23:
> Dependency on RPi.GPIO is a problem because RPi.GPIO does not support RPi5 and has problems with recent kernels. 

Agreed.

> [lgpio](https://abyz.me.uk/lg/py_lgpio.html) and [rpi-lgpio](https://github.com/waveform80/rpi-lgpio) (which emulates RPi.GPIO).

I'll take a look. I don't have an RPI5 though (so not high priority for me), but I'm sure you or someone else can verify a possible PR I might come up with, using an rpi5.

thijstriemstra, 2024-07-23:
Ah, it's a duplicate of #270

closed 2024-07-23 by thijstriemstra

Pako2, 2024-07-24:
Of course I read [#270](https://github.com/rm-hull/luma.core/issues/270), but I started a new issue anyway. I think it is not understood correctly. In the case of [#270](https://github.com/rm-hull/luma.core/issues/270), it's primarily that in some cases luma.core works without the RPi.GPIO library installed (and without any replacement for it), but based on the dependencies listed in _setup.cfg_, the system still tries to install it automatically and if that fails, then the entire installation fails.
On the contrary, in this case it is a matter of specifying the **lgpio** library instead of **RPi.GPIO** in the _setup.cfg_ (and of course the necessary changes must be made for this in the luma.core library). At the same time, I agree with [#270](https://github.com/rm-hull/luma.core/issues/270). Why is the dependency not optional (on any **RPi.GPIO** or **lgpio** type library)? After all, it is not even necessary for displays with an i2c interface.

Pako2, 2024-07-24:
And I have two more comments.
1. I don't have an RPi5 either and yet I ran into the problem. I created the [OLEDproc](https://forum.libreelec.tv/thread/25610-oledproc-add-on/?postID=169041) add-on for LibreELEC/Kodi about two years ago. In LibreELEC there is (perhaps from the very beginning of existence) also the "Raspberry Pi Tools" add-on, which also included the RPi.GPIO library. My addon also had the "Raspberry Pi Tools" addon listed as dependencies. However, with the new version of LibreELEC (LE12) there has been a change. The RPi.GPIO library was dropped so my [OLEDproc](https://forum.libreelec.tv/thread/25610-oledproc-add-on/?postID=169041) plugin stopped working.
2. The solution with **lgpio** plus **rpi-lgpio** works for RPi 4 (and should for RPi5 too) immediately, without changes in luma.core. But my wish is to completely remove the dependency on **rpi-lgpio** and that requires changes in luma.core.