rm-hull/luma.core #285: Use GPIO.setwarnings(False) warning

Trotter73, 2024-12-26:
Hi,

#### Type of Raspberry Pi
Pi Zero 2

#### Linux Kernel version
Linux test1 6.6.62+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.62-1+rpt1 (2024-11-25) armv7l GNU/Linux
Bookworm lite, 32it.
Luma has been installed with apt (python3-luma.lcd)

#### Expected behaviour
Santa brought me one of these for Christmas;
[DisplayHat Mini](https://shop.pimoroni.com/products/display-hat-mini?variant=39496084717651)

Basically its a ST7789 screen with a some buttons, I've got it working ok, with luma.lcd with one small exception, on the first run of a script I get the following warnings. Subsequent runs don't have it, I assume because the pins have been cleaned up... 

```
/usr/lib/python3/dist-packages/luma/core/interface/serial.py:206: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)
/usr/lib/python3/dist-packages/luma/core/interface/serial.py:357: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(self._gpio_CS, self._gpio.OUT, initial=self._gpio.LOW if self._cs_high else self._gpio.HIGH)
```

I think I understand why it's happening, pin13 hasn't been used and so has not has a state set, it is only a warning, really don't matter too much, and could be ignored but none the less I've been trying to get rid/supress it.

I came up with the following to supress the warning but was wondering if there was a more elegant way of doing it, any advice appreciated. Basically exercising pin 13 with warning suppressed before luma gets to it.....

Any advice appreciated.

```
from luma.core.interface.serial import gpio_cs_spi
from luma.core.render import canvas
from luma.lcd.device import st7789
import time
from PIL import Image, ImageDraw, ImageFont
import RPi.GPIO as GPIO

# Stop warnings on first run
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(13, GPIO.OUT)

serial = gpio_cs_spi(port=0, device=0, gpio_DC=9, gpio_CS=7, bus_speed_hz=52000000)
device = st7789(serial, width=320, height=240, rotate=2, gpio_LIGHT=13, active_low=True)

device.backlight(False)

# Load the specified font
font = ImageFont.truetype("dogicapixel.ttf", 8)  # Load the TrueType font

with canvas(device) as draw:
      draw.rectangle(device.bounding_box, outline="White", fill="black")
      draw.text((2, 2), "Hello World", font=font, fill="white")
time.sleep(3)
```

thijstriemstra, 2024-12-26:
see https://sourceforge.net/p/raspberry-gpio-python/wiki/BasicUsage/

```
GPIO.cleanup(13)
```

you could use python's `atexit` to run the cleanup when program exits.

Trotter73, 2024-12-26:
Hi,

Thanks for the quick reply, especially on Boxing day.

So I only see the warning, the first time the script is run after a reboot.

Boot -> Run script -> Warning
Run again -> No warning 

I see the same running the luma examples, get the warning on the first run of a script subsequent runs don't have it....