rm-hull/luma.core #286: RPi5: luma.core Fails with "Cannot determine SOC peripheral base address" Despite Working lgpio

DroidB, 2025-06-06:
Hello,

I am trying to use a ST7789 SPI display with luma.lcd on a Raspberry Pi 5, but the initialization fails with a RuntimeError: Cannot determine SOC peripheral base address. I have done extensive troubleshooting and believe this is a bug in how Luma.core interacts with the RPi5 environment.

My Setup:

Hardware: Raspberry Pi 5 Model B Rev 1.1 (Revision e04171)
OS: Raspberry Pi OS (Bookworm, 64-bit, fully updated June 2025)
Python: 3.11.2 (in a clean virtual environment)
Libraries: luma.core==2.4.2, luma.lcd==2.11.0, lgpio==0.2.2.0, Pillow==11.2.1
Summary of Findings:

The RPi.GPIO library (v0.7.1) fails on its own with the "Cannot determine SOC..." error, as expected for RPi5.
The lgpio library works perfectly standalone. I can open gpiochip0, claim pins, and write to them without issue.
The Adafruit-PlatformDetect library correctly identifies the board as a RASPBERRY_PI_5.
Even after uninstalling RPi.GPIO from the virtual environment (to force Luma to use lgpio), the luma.core.interface.serial.spi() constructor still fails inside its _configure method when it calls self._gpio.setup(). It raises RuntimeError: Cannot determine SOC peripheral base address.
This suggests that Luma's internal GPIO handler is either not successfully using the working lgpio backend, or Luma's own sysinfo or platform detection is failing in a way that produces this specific error.

This issue seems to be specific to Luma's initialization on the RPi5, as all the underlying tools (lgpio, spidev, platform detection) appear to be functional.

Thank you for your time and for maintaining this great library.

DroidB, 2025-06-07:
Hello,

I can confirm this issue still exists and I'd like to provide some detailed troubleshooting information. I have been working through this for hours and have definitively traced the problem.

My Setup:

Hardware: Raspberry Pi 5 Model B Rev 1.1 (Revision e04171)
OS: Raspberry Pi OS (Bookworm 64-bit, fresh install, fully updated as of June 2025)
Python: 3.11.2 (in a clean virtual environment)
Libraries: luma.core==2.4.2, luma.lcd==2.11.0
Summary of Findings:

I can confirm that the luma.core dependency on RPi.GPIO is the initial source of the Cannot determine SOC peripheral base address error on the RPi5.

I then followed the --no-deps workaround:

Created a clean virtual environment.
Installed luma.core and luma.lcd using --no-deps.
Manually installed Pillow, lgpio, spidev, and all other dependencies except for RPi.GPIO.
I was able to prove that the lgpio library (v0.2.2.0) works perfectly standalone in this environment and can control GPIO pins on the RPi5 without any errors.

However, even with RPi.GPIO uninstalled, the luma.core.interface.serial.spi class fails when trying to initialize an ST7789 display. The traceback shows it fails because the bitbang.__init__ method calls an internal function, __rpi_gpio__, which then unconditionally tries to import RPi.GPIO, leading to a ModuleNotFoundError: No module named 'RPi'.

This confirms that the --no-deps workaround is insufficient because the library code itself is hard-coded to call RPi.GPIO in this initialization path, rather than using the GpioManager() which would correctly fall back to the working lgpio library.

The fix will likely require changing luma.core.interface.serial.py to use from luma.core.gpio import gpio as GpioManager and calling GpioManager() instead of __rpi_gpio__().

Thank you for all your work on this project. I hope this detailed report helps in getting a fix out for RPi5 users.

rm-hull, 2025-06-07:
Does it need #284 merging and building .. sorry I'm a bit out of the loop with some of the state of the luma projects these days. @thijstriemstra any insights?

closed 2025-06-08 by DroidB

thijstriemstra, 2025-06-08:
> sorry I'm a bit out of the loop with some of the state of the luma projects these days. @thijstriemstra any insights?

I'm glad you asked Richard. I've been trying to help maintain the luma projects but your lack of response to my work has left me in a position where i'm unable to help out as long as current situation doesn't change. It costs a lot of time and patience to work on the project (for years) and not being able to move things forward because at every step I need to ask for changes that i can't make, like updating CI or asking for feedback or reviews that never happen, like you just pointed out. I feel I've shown that I can be trusted with these repo permissions in order to keep the project(s) alive but for some reason you don't want to do that or even acknowledge it so now i feel the projects are in limbo, which is very unfortunate seeing how much work was put in.